berarma/new-lg4ff #2: Cant use this on Fedora.

madbad, 2019-11-20:
`make install
make -C /lib/modules/`uname -r`/build M=$PWD modules_install
make[1]: ingresso nella directory "/usr/src/kernels/5.3.6-200.fc30.x86_64"
  INSTALL /home/madbad/Development/git/forcefeedback-driver/new-lg4ff/hid-logitech-new.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.3.6-200.fc30.x86_64
make[1]: uscita dalla directory "/usr/src/kernels/5.3.6-200.fc30.x86_64"
[root@localhost new-lg4ff]# depmod -A
[root@localhost new-lg4ff]# rmmod hid-logitech
rmmod: ERROR: Module hid_logitech is not currently loaded
[root@localhost new-lg4ff]# modprobe hid-logitech-new`
make install complain because the module cant be signed (but that should not be a problem).

The *hid-logitech* module for Fedora is built into the kernel (that's why it cant be unloaded)
This make not possible to use the new module (would require to rebuild the whole kernel)

Follow logs of when attaching the wheel
`dmesg | grep "logitec"
[  764.710291] logitech 0003:046D:C294.0002: input,hidraw1: USB HID v1.00 Joystick [Driving Force GT] on usb-0000:03:00.0-11/input0
[  765.398409] logitech 0003:046D:C29A.0003: input,hidraw1: USB HID v1.11 Joystick [Driving Force GT] on usb-0000:03:00.0-11/input0
[  765.398422] logitech 0003:046D:C29A.0003: Force feedback support for Logitech Gaming Wheels (new)
[  765.398423] logitech 0003:046D:C29A.0003: HZ (jiffies) = 1000, timer period = 4
[  765.726680] Error: Driver 'logitech' is already registered, aborting...
[  765.848867] Error: Driver 'logitech' is already registered, aborting...`

parkerlreed, 2019-11-20:
Yeah the SSL thing can be ignored. 

The Fedora kernel page lists HID Logitech as module and not built-in

https://src.fedoraproject.org/rpms/kernel/blob/master/f/configs/fedora/generic/CONFIG_HID_LOGITECH

You can check with 

```
zcat /proc/config.gz | grep -i hid_logitech
```

I've yet to be able to test this myself (On another distro though)

parkerlreed, 2019-11-20:
Hmm I wonder if 

```
CONFIG_LOGIWHEELS_FF=y
```

Could be the cause. Fedora and Arch both have that as built in. I may run into the same issue when I try later tonight. I'll let you know.

berarma, 2019-11-20:
The SSL error can be ignored. The new driver seems to be loaded looking at your dmesg output. The error about the driver being already registered can be ignored too. It's because there's two modules registered in the system for the same device. It looks like everything is OK. Have you checked the FF in the wheel after running these commands?

madbad, 2019-11-20:
> zcat /proc/config.gz | grep -i hid_logitech

```
[madbad@localhost ~]$ zcat /proc/config.gz | grep -i hid_logitech
gzip: /proc/config.gz: No such file or directory
```

parkerlreed, 2019-11-20:
@madbad Ahh ok was just checking. That config exposure is a setting at compile time. Kinda annoying they disable that.

@berarma Thanks for the reassurance. Can't wait to test this out tonight.

berarma, 2019-11-20:
> Hmm I wonder if
> 
> ```
> CONFIG_LOGIWHEELS_FF=y
> ```
> 
> Could be the cause. Fedora and Arch both have that as built in. I may run into the same issue when I try later tonight. I'll let you know.

This has to be set to "y" for the module to be built with driving wheels support. It's not about the module itself.

parkerlreed, 2019-11-20:
@berarma Cheers. I hope you liked the small gift on Reddit the other day ;)

berarma, 2019-11-20:
@parkerlreed, I haven't seen any. Maybe you rewarded Leillo, he posted the news. If that's the case it would be a very well spent gift, his support in this project has been very important. Cheers.

parkerlreed, 2019-11-20:
@berarma I awarded his main post and your user directly. :smile: Honestly, thanks a ton for the work that has gone into this. My G29 has been sitting unused due to the sheer lack of support in Linux titles. Hopefully this fixes most if not all of that.

madbad, 2019-11-20:
Ok, gave a quick try on the native ETS2 game and it seems to be working.
Will give a more proper test with a wine rFactor2 iinstall tomorrow
(I'm using a DFGT)
Thanks for the help guys

berarma, 2019-11-21:
@madbad, ETS2 needs a fix in the direction of the effects. It applies some effects in the vertical axis but driving wheels use only the horizontal axis. The https://github.com/berarma/ffbtools package has the command "ffbwrap" that allows to apply a workaround. Put this line in the steam launch options:

`ffbwrap --direction-fix /dev/input/<your_wheel> -- %command%`

You should notice a very strong rumbling of the motor.

This is a problem in some Wine/Proton games too.

parkerlreed, 2019-11-21:
@berarma Just about to test this. I've never even thought about it, but what's the best mode for the G29 to be in? PS3 or PS4?

parkerlreed, 2019-11-21:
Answered my own question... FF doesn't work at all in the PS3 mode. PS4 it is

leillo1975, 2019-11-21:
????
FFB on PC's uses PS3 mode

berarma, 2019-11-21:
> Answered my own question... FF doesn't work at all in the PS3 mode. PS4 it is

That's weird, it's supposed to work in PS3 mode. I haven't tried it in PS4 mode. Please, post your dmesg output.

parkerlreed, 2019-11-21:
@berarma yeah as noted elsewhere the PS4 mode doesn't work. I was confusing the PS4 mode auto centering itself as the driver doing it not realizing it just does that itself. 

But we're all good now. I'm just slowly trying to go through all the games I have and see what is supported. Thanks again for the help.

berarma, 2019-11-21:
I've tried to improve the build and install steps based on the feedback here.

closed 2019-11-21 by berarma

parkerlreed, 2019-11-21:
@berarma Is it normal for the PS3 mode calibration to go all the way right, then left, but not spin back to center? I'm assuming this is just in hardware? It finds the center correctly as evident by Oversteer launching.

berarma, 2019-11-21:
The calibration process is entirely done by the firmware. Sometimes it spins back at the center, sometimes it misses the center, and sometimes it doesn't spin back. I'm not sure why this happens but I haven't had any problem, the wheel is always perfectly calibrated.

It seems like the firmware applies autocentering at full force for a very short and undefined time when the autocalibration ends, but it doesn't wait for the wheel to get there.