berarma/new-lg4ff #3: Fails to compile on 5.4rc7

parkerlreed, 2019-11-21:
xpad also has this issue and I reported it there. Seems something was removed in master.

I'm currently on master with a patch as they broke my soundcard with some new usb descriptor validation... I'll have to compile the distro 5.3.11 with the same patch to be able to compile this project. Just wanted to give you a heads up.

```
[parker@e55 new-lg4ff]$ make -j10
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/5.4.0-rc7-g96b95eff4a59/build'
  AR      /home/parker/build/new-lg4ff/built-in.a
  CC [M]  /home/parker/build/new-lg4ff/hid-lgff.o
  CC [M]  /home/parker/build/new-lg4ff/hid-lg.o
  CC [M]  /home/parker/build/new-lg4ff/hid-lg3ff.o
  CC [M]  /home/parker/build/new-lg4ff/hid-lg4ff.o
  CC [M]  /home/parker/build/new-lg4ff/hid-lg2ff.o
In file included from ./include/linux/power_supply.h:17,
                 from ./include/linux/hid.h:27,
                 from /home/parker/build/new-lg4ff/hid-lgff.c:21:
./include/linux/leds.h:11:10: fatal error: dt-bindings/leds/common.h: No such file or directory
   11 | #include <dt-bindings/leds/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/parker/build/new-lg4ff/hid-lgff.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ./include/linux/power_supply.h:17,
                 from ./include/linux/hid.h:27,
                 from /home/parker/build/new-lg4ff/hid-lg3ff.c:13:
./include/linux/leds.h:11:10: fatal error: dt-bindings/leds/common.h: No such file or directory
   11 | #include <dt-bindings/leds/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/parker/build/new-lg4ff/hid-lg3ff.o] Error 1
In file included from ./include/linux/power_supply.h:17,
                 from ./include/linux/hid.h:27,
                 from /home/parker/build/new-lg4ff/hid-lg2ff.c:14:
./include/linux/leds.h:11:10: fatal error: dt-bindings/leds/common.h: No such file or directory
   11 | #include <dt-bindings/leds/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from ./include/linux/power_supply.h:17,
                 from ./include/linux/hid.h:27,
                 from /home/parker/build/new-lg4ff/hid-lg.c:17:
./include/linux/leds.h:11:10: fatal error: dt-bindings/leds/common.h: No such file or directory
   11 | #include <dt-bindings/leds/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/parker/build/new-lg4ff/hid-lg2ff.o] Error 1
make[2]: *** [scripts/Makefile.build:266: /home/parker/build/new-lg4ff/hid-lg.o] Error 1
In file included from ./include/linux/power_supply.h:17,
                 from ./include/linux/hid.h:27,
                 from /home/parker/build/new-lg4ff/hid-lg4ff.c:15:
./include/linux/leds.h:11:10: fatal error: dt-bindings/leds/common.h: No such file or directory
   11 | #include <dt-bindings/leds/common.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:266: /home/parker/build/new-lg4ff/hid-lg4ff.o] Error 1
make[1]: *** [Makefile:1652: /home/parker/build/new-lg4ff] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.4.0-rc7-g96b95eff4a59/build'
make: *** [Makefile:9: default] Error 2
```

parkerlreed, 2019-11-21:
I just realized the file is still there. I think the package I used to build it just didn't copy it over in the headers. I'll close out for now until I can figure out if this is upstream or just packaging error.

closed 2019-11-21 by parkerlreed

berarma, 2019-11-21:
This could be an upstream problem if it happens with a pristine kernel source. We don't use this file, it gets included by another kernel file.