berarma/new-lg4ff #4: I'm still lost

parkerlreed, 2019-11-21:
G29. Even with the new driver, I'm having the worst time trying to get this working ANYWHERE

Proton doesn't see it

Euro Truck2 native doesn't see it despite it being available and working 

https://i.imgur.com/09CtyGt.png

What am I doing wrong? 

PS3 mode the driver attaches, the wheel does the spinning initialization but doesn't center by default. I can go into oversteer and set the rotation and autocenter and it works.

PS4 mode initializes with centering but the driver doesn't attach to that.

I'm just as lost as ever trying to get this to work in anything useful.

parkerlreed, 2019-11-21:
Oh so for Euro Sim "It just works" 

Even though it doesn't show the controller if I do keyboard + wheel it works when driving. I guess Ill see if I have anything else to test.

closed 2019-11-21 by parkerlreed

berarma, 2019-11-21:
I'm sorry you had problems. This game has been tested by me and others with the G29 and we didn't find any issue. We use PS3 mode.

I did have to use a hack provided by ffbtools to change the direction of the effects or some of the effects wouldn't play.