berarma/new-lg4ff #5: I can't install on Manjaro

franciscot, 2019-11-21:
I installed linux419-headers and kmod.
I get that error on manjaro trying sudo make install:

```
make -C /lib/modules/`uname -r`/build M=$PWD modules_install
make[1]: se entra en el directorio '/usr/lib/modules/4.19.84-1-MANJARO/build'
  INSTALL /home/X/new-lg4ff/new-lg4ff/hid-logitech-new.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:69
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:76
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.19.84-1-MANJARO
make[1]: se sale del directorio '/usr/lib/modules/4.19.84-1-MANJARO/build'
```
Any idea?

Thanks.

jotomo, 2019-11-21:
Those messages didn't abort the build. Check with ```ls -hs /lib/modules/`uname -r`/extra```. The module should be there. I got the same messages but the module was build and works.

berarma, 2019-11-21:
The install process tries to sign the new module and fails. It happens on some systems. This step isn't needed and the errors are harmless.

closed 2019-11-21 by berarma