berarma/new-lg4ff #6: Feedback for Dirt Rally 2.0 with Logitch G27

jotomo, 2019-11-21:
Name of the game: DIRT Rally 2
Steam ID: 690790
Game version: 1.10.1
Game Type (native/proton/wine): Proton
Proton/Wine version: 4.11-8
Wheel Model: Logitech G27
Compatibility Mode: None
Distribution: Arch Linux
Kernel version: 5.3.11-zen1-1-zen

Previously I've using an older proton build with SDL-Joystick support disabled to work around an issue in proton to make FFB work. With that and the normal kernel _hid-logitech_ module, FFB worked in game when replugging the wheel after game start (as a side-effect the wheel showed up twice, causing funky things when not mapping buttons the same on both). This was without _ffbwrap_.

I've switched now to proton 4.11-8, using _ffbwrap_ with _--update-fix_ and _hid-logitch-new_ module and compared FFB on Windows and Linux, activating one game FFB setting (wheel friction, suspension etc) at a time to compare. The result is that it's the same but for one effect, wheel friction: with the car stationary (easiest to produce the effect), turning the wheel quickly gives a constant force against the turning motion. Turning at a slower speed the motors seem to try to create the friction by changing direction at high frequency. This makes the wheel rather stiff, while sending a lot of small vibrations through the mounting while also being rather loud. 
I've seen this effect elsewhere (I think it was Project Cars 2 on Windows), but never with DR2.
So overall it works rather well, except for the issue mentioned (hopefully explained in an understandable way).
If that issue is considered an error (functionally it works, but the noise makes me wonder if I'll wake up anybody if drive late) I'd produce logs (with both setups) to help with debugging.

berarma, 2019-11-21:
That's very interesting. I'd like to see a log of the friction effect producing the vibrations. The friction effect isn't properly supported by the hardware, it produces a damper effect instead (I've been told it's the same on Windows). Most probably it's being emulated by the game using constant force.

I like your testing method. We want to implement the best support possible, so I consider any difference with the Windows driver an error, unless it's for good.

Thank you very much.

jotomo, 2019-11-22:
Here's the log: 
[dr2.ffbwrap.log-20191122225631.log](https://github.com/berarma/new-lg4ff/files/3881585/dr2.ffbwrap.log-20191122225631.log)
I started the game, disabled all FFB but wheel friction, stood on the starting line turning wheel at multiple speeds before exiting again.
Maybe two short clips help demonstrate the issue better than me trying to explain it;
With Windows: https://vimeo.com/375016869 
With Linux https://vimeo.com/375016613
When turning with medium speed, the update frequency(?) creates an interesting resonance(??).

berarma, 2019-11-23:
I'll try to reproduce the problem and debug it. Thank you very much for the detailed information.

berarma, 2019-11-25:
Looking at the videos I wonder if you really get a friction effect or a damper effect instead. The friction effect feels like more resistance when moving the wheel slowly and less resistance when moving it fast. The damper effect feels like more resistance the faster the wheel moves. Which effect do you get?

My G29 plays the friction effect like a damper effect, there's no difference. I'm failing to reproduce your problem and it might be because we are using different wheel models.

jotomo, 2019-11-25:
I think I'm getting a damper effect - more resistance when turning faster.
I've tried your last commit, with it the undesired noise occurred much more significantly. So instead of the multiplication done by the patch I divided by 2, which reduced the noise to an insignificant bit. With a division of 3 (experimenting :-) ) it vanished completely, but the wheel forces felt a bit less strong.
Lastly, I joined the case branches for FF_FRICTION & FF_DAMPER, so that friction was played as FF_DAMPER, like it does for your G29 and the wheel felt pretty much exactly like on Windows, so I'd guess friction is always played as damper effect as well on the G27 (at least with DR2).

jotomo, 2019-11-25:
Did some more comparing, there is no discernible difference between playing FF_FRICTION with a hacked divisor compared to just playing it as FF_DAMPER effect.

howlett, 2019-11-29:
It might be worth retesting with Proton 4.11-9.  The change log has this: 
"Restore steering wheel controller force feedback."

Full changelog here: https://github.com/ValveSoftware/Proton/wiki/Changelog

jotomo, 2019-11-29:
@howlett I see no difference with Proton 4.11-9: I still need _ffbwrap_ with the update fix.
However, in testing again I'm learning to compare better (even with a script automating hibernate-rebooting into windows and starting DR2, comparing nuances is a bit tricky with the delay in between).
I'd now say that I can differentiate between the friction and damper effect.
With current new-lg4ff I do get a friction effect (noticeable resistance right away at low turning speeds; compared to almost no resistance when turning slowly and gradually increasing resistance with the damper effect). And I would say I get the same friction effect on windows.
To sum up (and hopefully be more clear): the G27 does support the friction effect (and it feels better with it, as it smooths out and stabilizes the wheel). With current new-lg4ff the effect causes the described rattling effect (more so with the latest patch), that I have locally mapped friction to be played as damper as a workaround.
Not sure where to go from here, but I'm terribly excited about how far this has come along and I'm very thankful @berarma for your work!

berarma, 2019-12-01:
@jotomo, please try loading the module with different options. Try with "fixed_loop=1", then "timer_msecs=8" and also with both options. Also, check with dmesg for module messages saying something like "queue full".

jotomo, 2019-12-01:
@berarma No change with timer 2-16ms, with either fixed loop or as fast as possible. No messages with `dmsg`.

The G27 does ineeed support the friction effect (from the LG FFB v1.5 protocol document):
```
This force is only supported on the following devices:
 Logitech Driving Force Pro (PID_C298)
 Logitech G25 Racing Wheel (PID_C299)
 Logitech Driving Force GT (PID_C29A)
 Logitech G27 Racing Wheel (PID_C29B)
```
The mapping into the submitted structure (_report_) looks good (as described in the document).
Unfortunately for that effect the valid range for K1 and K2 parameters isn't defined, or I haven't yet found the correct table it uses.
I'll see if I can figure out what range is correct. Reducing K1/K2 significantly reduces the rattling, while the effect can still be pretty strong if turned up in the game (or put differently: strength does up, rattling goes down, so that seems like a step in the right direction).

berarma, 2019-12-01:
By the lack of information I'm guessing it should use all 8 bits in a linear function as all other conditional effects do. I think the G29 model didn't exist when the document was written and that's why it isn't mentioned anywhere. It's basically a G27 with external changes so it should behave the same. I'd like to confirm all this because reports are contradicting.

Possible the only way to know for sure what's happening would be to use a USB HID inspector while sending min/med/max level effects to the wheel on Windows. Maybe they're using a mapping function customized per wheel model. I'm sure the released doc doesn't tell us everything and is probably outdated.

I'm working on a basic testing tool that will hopefully help somewhat.

berarma, 2019-12-25:
There's a new version out with improvements that might affect this issue. It would be interesting to see if this is still reproducible with the latest version. Thanks.

jotomo, 2019-12-26:
I see now change with 0.2a.

berarma, 2019-12-27:
There must be some quirk with the G27. I don't have a G27 to try and my G29 seems to work similar to what you're describing on Windows. For me it still plays like a damper but it seems to boost sooner and has more strength.

I'm thinking on contacting someone at Logitech to try and get an updated specification. The other option would be someone figuring out how the G27 works through reverse engineering.

jotomo, 2019-12-28:
If there's a way of reaching a technical contact at Logitech that'd be great. I'm afraid reverse engineering is currently not an option for me - too many software projects on my plate already.

berarma, 2020-01-20:
I've been learning about how clipping affects Logitech wheels. Since each effect slot (there are four) can use the full force range, the four slots playing full force effects at the same time would cause lots of clipping. It's like what happens in an audio mixing desk, you have to setup the mix in a way that doesn't saturate the output but the result is still as powerful as possible..

I recalled your comment about the friction effect being stronger on new-lg4ff than it was on Windows, and I've also read somehere that the Logitech Windows driver had settings to tone down the spring, damper effects, and maybe the friction effect too. It makes sense.

Conditional effects are less important than constant/sine/etc effects. They don't give the most important information about your driving. And in most games they're static. So it makes sense to tone down these effects to leave more room for the other ones.

I've added settings to setup the gain of these effects. It can be found in the master branch. I've updated https://github.com/berarma/oversteer too so that they can be easily changed. The default gain for spring, damper and friction effects is now set at 30%. I think it's the default value on the Windows drivers too. It's possible to disable them completely to allow the other effects to use the full range, and also turning them up for games that make good use of them.

There's also some features added for monitoring FFB clipping

I hope this is an improvement. I'd like to know.

jotomo, 2020-01-20:
@berarma  Awesome update, DR2 has just become perfect for me on Linux :-) Thank you so much, again!

- I've compared Tire Friction and Wheel Friction (in-game options) each as the only enabled FFB setting (which play as, or at least feel like, damper and friction effect respectively). They feel the same on Win7 and Linux. Combined with all effects it feels overall better and complete now.
- The auto-center at the start of a stage (if the wheel was not close to center) previously exhibited the rattling issue. That is now gone and the wheel turns smoothly to center. No rattling while driving either.
- Wheel friction strength (as the only enabled FFB setting) had about the same (_maybe_ a bit less) strength compared to Win7, only when turning up above 100 (percent presumably, no unit given in-game) to 150 the strength went down to say 90%. 
I might need to recheck the last point on another day - nuances get harder to detect the longer I wrangle the wheel with different settings.
- Oversteer says overal force feedback gain is at 100, autocenter was at 0 (default), others are greyed out for my wheel. I think the game overrides those though.

Let me know if you need more tests done or logs provided.

berarma, 2020-01-20:
I'm glad it has worked so well. :)

> Oversteer says overal force feedback gain is at 100, autocenter was at 0 (default), others are greyed out for my wheel. I think the game overrides those though.

It's greyed out because you don't have the permissions to change those settings. Oversteer should warn about this at startup but there was a bug that made the check fail. Please, update and try again.

These settings are important. They're not overriden by the game, they multiply the ones in the game. If you have friction level set to 30% and in the game at 50%, you get 15% of the effect. So the possible variance you've found in the friction effect might be because the default 30% value in the driver isn't the same multiplier that the Windows driver uses. Tweaking these values you should be able to mimic the Windows driver unless there's something else still missing.

Please, if you test it and find values that you think give an output really close to the Windows driver report back and I might update the driver to those values.

Thank you very much for your helpful reports and support!

jotomo, 2020-01-22:
Did a bit of testing. With levels (spring, damper, friction) all adjusted in parallel, with all in-game settings at 100. 40 is definitely too much. At 35 I get minor rattling, at 33 only hints of rattling are left. 32 feels as strong as Windows. 30 seems a good conservative default with a bit of headroom to spare, though 32 would also work to be closer to Windows. Of course I can only make this statement about the G27 combined with DR2. I'll continue with 32 and will report back should any issues occur with more time spend driving.

berarma, 2020-01-22:
I'm surprised you can notice such small changes. From 30 to 32 that's a really small difference. I was thinking there could be a bigger mismatch in some of the effects.

Please, keep me updated on your testing.

jotomo, 2020-01-22:
> I'm surprised you can notice such small changes. From 30 to 32 that's a really small difference.

I wanted to properly test it :-) But yes, the differences are very small to the point where it might cross over into just thinking there must be a difference because a setting was changed (which is why I started from 100 -> 60 -> 40, to make sure there are significant differences to get a good baseline). Should be a good point to call it good enough for that test.

> I was thinking there could be a bigger mismatch in some of the effects.

Since I tweaked all levels at once: would testing those separately (all zero but one setting) yield any tangible benefit? And if so, is there documentation how to interpret the FFBmeter LEDs in oversteer (they work nicely with the G27 btw)? I guess checking if clipping occurs only makes sense with multiple effects active...

berarma, 2020-01-22:
The purpose of the FFBmeter is to adjust the FFB in the games to avoid significative clipping. It only makes sense when testing with all the effects that you're going to use in that game. It's not useful outside the context of the game and/or car you're going to use it on.

When all leds are on, the FFB is maxed out or almost, when all leds are off the FFB is off or too weak to be noticed. When one green led turns off there's some clipping but it's still acceptable, probably almost unnoticeable. When two green leds turn off there's maybe a bit too much clipping but maybe still acceptable. The more leds turn off from left to right the more clipping is happening.

Since Logitech wheels don't have lots of force, a small amount of clipping is desirable (one or two green leds turning off). Even more clipping is acceptable when the car crashes or drives over very rough ground because there's not much detail to notice in these situations.

I'll write all this info in the README before making a new release.

azampatti, 2020-01-30:
(OFF TOPIC, didn't want to open a new case for this) 
Guys, my old  G25 (11 years old!!) started to fail on me. Even if I could bring it back to life, i got a pretty good deal in an open box G29 which I've set up today and works just like it should.... except one thing.

I couldn't get the shift lights to work. Tried on my Windows Laptop and they work out of the box as expected. Did you see this issue? is it known?  100% not a big deal but I'm curious :)

thanks!

jotomo, 2020-02-11:
@berarma Everything has been running smoothly for the last 3 weeks (including the 0.3 release, oversteer from master, tweaking FF strengths, using the LED feedback etc). Nothing to report other than that I continue to enjoy your work tremendously :-). Hence, I'll close this as done. I'll create a new issue should I have any other useful feedback. If you think I can help you with testing in any way, please ping me anytime.

closed 2020-02-11 by jotomo

berarma, 2020-02-11:
Thanks @jotomo for your support on this and other issues!! Your findings in your detailed FF tests made me realize what was still missing.