berarma/new-lg4ff #7: fftest doesn't do anything

parkerlreed, 2019-11-22:
All the effects say they uploaded fine but there's no response selecting any of them. Is there any way to test the effects in general?

```
[parker@e55 bin]$ fftest /dev/input/event20
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event20 opened
Features:
  * Absolute axes: X, Y, Z, RZ, Hat 0 X, Hat 0 Y, 
    [27 00 03 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Constant, Periodic, Ramp, Spring, Friction, Damper, Rumble, Gain, Autocenter, 
    Force feedback periodic effects: Square, Triangle, Sine, Saw up, Saw down, 
    [00 00 00 00 00 00 00 00 00 00 BF 1F 03 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
Enter effect number, -1 to exit
4
Now Playing: Strong Rumble
Enter effect number, -1 to exit
5
Now Playing: Weak Rumble
Enter effect number, -1 to exit
^C
```

parkerlreed, 2019-11-22:
Oh I see https://github.com/berarma/ffbtools has ffbplay. I assume fftest is just older and doesn't know what to send?

berarma, 2019-11-23:
The fftest command is older but it uses the same API and it's totally compatible. It uses a strange mix of parameters to play the effects though.

And that's why it has helped us unveil a bug where effects didn't play if the global gain setting wasn't at the max level. This doesn't normally happen because most applications leave it at the maximum setting.

closed 2019-11-23 by berarma