berarma/new-lg4ff #10: Meaning of the direction field in the effects struct

berarma, 2019-11-23:
This is a call for anyone that can share any info to understand how the direction field should be interpreted for driving wheel devices.

The Linux FF API provides a field to specify and angle at which to play an effect, 0º would mean down, 90º left, 135º is left and up. Wheels are supposed to be able to play FF effects only by rotating the wheel in one direction or the other.  So this direction value gets decomposed by the kernel driver into its horizontal and vertical axis values and only the horizontal value is used. This means that an effect with a vertical direction will never be played.

Still, some games send effects with a vertical direction that never get played. I don't understand why any game would send such effect if it's not going to be played. Can someone explain why this might be happening?

ETS2 does this, and other games running with Proton do too.

berarma, 2019-11-24:
@simon50keda

berarma, 2019-11-25:
https://docs.microsoft.com/en-us/previous-versions/windows/desktop/ee417536(v=vs.85)?redirectedfrom=MSDN

simon50keda, 2019-11-26:
Well I don't have any definitive or right answer, as I was asking myself same question back then, if I remember correctly. However it could be is that Y axis could be other input accepting FFB?

CKWarner, 2019-11-30:
Bear in mind that I'm not a dev of any kind, nor do I have experience in this area, but, from the link you provided:

> The direction of an effect is the direction from which it comes. An effect with a direction along the negative y-axis tends to push the stick along the positive y-axis (toward the user). If the user must push the stick toward the left to counteract an effect, the effect has a left direction; that is, it lies on the negative x-axis.
> Direction can be expressed in polar, spherical, or Cartesian coordinates.
> Polar coordinates are expressed as a single angle, in hundredths of a degree clockwise from whatever zero-point, or true north, has been established for the effect. Normally this is the negative y-axis; that is, away from the user. Thus an effect with a polar coordinate of 9,000 has a direction of east, or to the user's right, and the user must exert force to the right to counteract it.

So the coordinate axis for the effect isn't fixed: it could have been defined relative to whatever direction makes sense for that particular effect at that particular time. Also you can see that, for example, forces that act to straighten the wheels would be generally useful.

You could also ask SCS why they do the things they do; my impression is that they're interested in having better force-feedback support on Linux but don't have the resources or inclination to do the kind of untangling that you're trying to do.

berarma, 2019-12-01:
There's been some progress on this. I was trying the ETS2 demo that has awful FFB, the full game has much better FFB and solves any problems the demo had. We're also closer to solving the problems in Proton. It seems some bugs were introduced while porting the effects to SDL2.

So, at this moment it seems like there's nothing weird to worry about, just bugs. I'll close this issue when it's all confirmed and good.

berarma, 2019-12-02:
It seems this can be already closed. There was nothing more than bugs. Thanks for your support!

closed 2019-12-02 by berarma