berarma/new-lg4ff #11: What is the purpose of checking if KERNELRELEASE is blank in the Makefile?

parkerlreed, 2019-11-25:
I'm attempting to add dkms support and when it passes in KERNELRELEASE, the first half of the if condition is met in the Makefile and thus fails to build.

parkerlreed, 2019-11-25:
Well before making a pull request, here's my changes. Removed the if/else statement and added dkms as a target. Builds fine both locally in the tree and through dkms

https://github.com/parkerlreed/new-lg4ff/commit/6026e683085bc221c8103a909bc44ce0f0af8860

berarma, 2019-11-25:
That's cool!

Don't be afraid to create a PR now. We can comment changes there and make corrections all in one place before merging.

I followed instructions in section 3.1 from [this document](https://www.kernel.org/doc/Documentation/kbuild/modules.txt). If that's an issue for DKMS support then I would use a Kbuild file as explained in section 3.2.

I don't think we should manage the module source code location from the Makefile. I'd rather change the installation instructions so that users install it at /usr/src and use dkms directly. I guess packagers will find it easier this way too.

So, I'm thinking we just need the `dkms.conf` file and an updated README. What do you think?

parkerlreed, 2019-11-25:
Yeah that would work too. If I can't remember how to squash commits I may nuke and redo it :P

Also if you have a better idea for version number, let me know. I just picked that as it was simple and there was nothing to reference.

berarma, 2019-11-26:
PR #13 created.

closed 2019-11-26 by berarma

parkerlreed, 2019-11-26:
@berarma https://aur.archlinux.org/packages/new-lg4ff-dkms-git/ :)