berarma/new-lg4ff #15: Reseting on every startup

Linoslav, 2019-11-28:
Hi! I've been using 50keda pluging FFB in ETS2 and ATS and this is far more convinient. Also, great work with adding more game inputs like engine rumble (great in trucking sim to have tactile info without looking into dash). Thanks for your work. I just have two small issues:
1st I need to reset everything every time I start computer (unload then load)
2nd Data when "centering" wheel is presented as little -kicks- that are stronger with the turning and I believe that Feral Interactive had approached the same issue by making it harder (more resistive) to turn the wheel when more offset

Also, I don't know if this made the Wreckfest getting the FFB working with G29 or they made some adjustments in game (haven't played a while), but seems to help in that game also. Once again, I know that Logitech made a mess with this model and thank you for your work.

berarma, 2019-11-29:
Hi Linoslav.

It makes me glad that people are enjoying it.

I don't know how 50keda's plugin works but with this driver I'm just playing the effects the game sends, not creating my own.

1. This can be solved installing the driver using DKMS. There a instructions in the README. Have yout tried? Feedback welcome.

2. I guess you're refering to the SPRING effect. The issue you describe it's related to the mechanics of the wheel, low forces don't create a smooth movement. I could apply corrections to the game's inputs but I'm not sure how it should be done to really improve the effects for all games, not just one. Turning up the Spring effect would probably produce clipping on some other situation and that's bad. I don't know if Logitech applies some custom curve function in their Windows driver to improve the dynamics of the wheel, we'd need to do measured tests to know.

Linoslav, 2019-12-02:
Hi again!
I will try to install it, but it's not a problem to hit two lines of code either way. After spending several hours using it, I noticed what's the issue here. When turning wheel (like in curves or whatever) effects seem to multiply each other. So I don't feel kicks but rather other "effects" turn up.
And it would be great if reaction to terrain can be toned down (like less movement in wheel or something), I've put it on minimum and it still kicks the wheel out of hands when hitting the smallest bump (though I heard that is a thing with Logitech in Linux).
P.S. I am just giving feedback to your great work, not in engineering or development way, but just as user wanting to help. This is great work, what you have made is much more than companies do with all their resources. Thank you!

berarma, 2019-12-02:
I play the game with a G29 wheel and it works pretty well after tunning the effects. I would recommend toning down the collissions effect that I think it's used when hitting curbs or other obstacles big and small. You can try all sliders one by one until you find the one that removes the hard kicks.

I would also try loading the module with option "timer_msecs=8" or "timer_msecs=12" in case there are "queue full" messages in dmesg. These games may throw effect updates without any timing control, at least the demos could do it. They seem to have largely improved FFB in the game since the demos were released but still may be useful.

Linoslav, 2019-12-02:
I didn't even think of that. Collision is set on middle. I will adjust. Didn't really mind at all, I rarely have collisions :D Great, thanks!

berarma, 2019-12-02:
Please, report your findings. If there's really a problem I'd like to know. Thanks.

closed 2019-12-25 by berarma