berarma/new-lg4ff #17: Couldn't get to install on newest arch's kernels

Marioh2451, 2019-12-03:
I used the installation from dnmodder, it has installed successfully until it went to the DKMS installation. It works but doesn't get it to work with arch kernel 5.3.12-arch-1-1 why is it asking for older kernel headers when I have the newest?

parkerlreed, 2019-12-03:
Did you rebuild the package? You would have to update if using an older package

The package I created in the AUR successfully updated when going to 5.4

```
( 8/18) Install DKMS modules
==> dkms install new/lg4ff -k 5.4.1-arch1-1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=5.4.1-arch1-1 KVERSION=5.4.1-arch1-1....
cleaning build area...(bad exit status: 2)
Kernel cleanup unnecessary for this kernel.  Skipping...

DKMS: build completed.

hid-logitech.ko.xz:
Running module version sanity check.
 - Original module
   - Found /usr/lib/modules/5.4.1-arch1-1/kernel/drivers/hid/hid-logitech.ko.xz
   - Storing in /var/lib/dkms/new/original_module/5.4.1-arch1-1/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /usr/lib/modules/5.4.1-arch1-1/kernel/drivers/hid/

depmod....

DKMS: install completed.
```

parkerlreed, 2019-12-03:
Also who's dnmodder?

berarma, 2019-12-04:
@Marioh2451, please, show us all the commands you run and their output.

You can run the command `script new-lg4ff.log` in a console, then repeat every step from the start, when finished run the `exit` command. Then please attach the file `new-lg4ff.log` to this issue.

Marioh2451, 2019-12-05:
Construyendo new-lg4ff-dkms-git...
==> Creando el paquete: new-lg4ff-dkms-git r28.e7601c0-1 (jue 05 dic 2019 21:50:21 CET)
==> Comprobando dependencias mientras se ejecuta...
==> Comprobando dependencias mientras se compila...
==> Recibiendo las fuentes...
  -> Actualizando new-lg4ff del repositorio git...
Extrayendo origin
==> Validando los archivos source con md5sums...
    new-lg4ff ... Omitido
==> Eliminando el directorio $srcdir/...
==> Extrayendo las fuentes...
  -> Creando copia de trabajo de new-lg4ff del repositorio git...
Clonando en 'new-lg4ff'...
hecho.
==> Iniciando pkgver()...
==> Entrando en entorno fakeroot...
==> Iniciando package()...
==> Depurando la instalación...
  -> Quitando los archivos libtool...
  -> Purgando los archivos innecesarios...
  -> Quitando los archivos de las bibliotecas estáticas...
  -> Despojando los símbolos innecesarios de los binarios y de las bibliotecas...
  -> Comprimiendo las páginas del manual y de información...
==> Buscando problemas de empaquetado...
==> Creando el paquete «new-lg4ff-dkms-git»...
  -> Generando el archivo .PKGINFO...
  -> Generando el archivo .BUILDINFO...
  -> Añadiendo el archivo install...
  -> Generando el archivo .MTREE...
  -> Comprimiendo el paquete...
==> Abandonando el entorno fakeroot.
==> Compilación terminada: new-lg4ff-dkms-git r28.e7601c0-1 (jue 05 dic 2019 21:50:23 CET)
==> Limpiando...

Preparing...
Resolving dependencies...
Checking inter-conflicts...
Running post-transaction hooks...
Checking keyring...
Checking integrity...
Loading packages files...
Checking file conflicts...
Checking available disk space...
Installing new-lg4ff-dkms-git (r28.e7601c0-1)...
Running post-transaction hooks...
Arming ConditionNeedsUpdate...
Install DKMS modules
==> dkms install new/lg4ff -k 5.4.1-arch1-1
==> Unable to install module new/lg4ff for kernel 5.3.12-arch1-1: Missing kernel headers.
Operación finalizada con éxito.

Okay this is the one i used from aur, it didn't work, so what I have done is removing the package I used the manual one. Now after looking at this github repo, i have found out the new "dkms installation" tutorial where i can use this. after that, I wrote the necessary commands to do the installation and finally after long and long time I got this to work.

closed 2019-12-05 by Marioh2451

berarma, 2019-12-05:
@parkerlreed, could we me mention your Arch package in the README? Is it ready for general use?

parkerlreed, 2019-12-05:
I mean it needs some work. Pacman *should* automatically handle running dkms remove on package uninstallation (alpm hooks in the dkms package) but for some reason it just bails with no error (even after modifying the scripts to be more verbose)

I've taken a couple stabs at it now but haven't gotten anywhere with it. So I just manually run the dkms remove on uninstall via .install file. I'd say it works but it may need some attention.

berarma, 2019-12-05:
[dnmodder](https://aur.archlinux.org/packages/?K=dnmodder&SeB=m) is the user that packaged [Oversteer](https://github.com/berarma/oversteer) for Arch. He was trying to package this project until he found out you already did it. You might be able to help each other.

parkerlreed, 2019-12-06:
Ahh thank you!

parkerlreed, 2019-12-06:
Added him as a co-maintainer https://aur.archlinux.org/packages/new-lg4ff-dkms-git/#comment-719521

I'd look into this more but I've got other stuff I'm fighting with at the moment.