berarma/new-lg4ff #18: Module is named "hid-logitech" instead of "hid-logitech-new" on Ubuntu

Patola, 2019-12-15:
```
[19:20] [root@risadinha dkms]# dkms install /usr/src/new-lg4ff

Creating symlink /var/lib/dkms/new-lg4ff/0.1/source ->
                 /usr/src/new-lg4ff-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j16 KERNELRELEASE=5.0.0-23-lowlatency KVERSION=5.0.0-23-lowlatency...
Signing module:
 - /var/lib/dkms/new-lg4ff/0.1/5.0.0-23-lowlatency/x86_64/module/hid-logitech.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.

hid-logitech.ko:
Running module version sanity check.
 - Original module
 - Multiple same named modules!
   - 2 named hid-logitech.ko in /lib/modules/5.0.0-23-lowlatency/
 - Installation
   - Installing to /lib/modules/5.0.0-23-lowlatency/updates/dkms/

depmod...

DKMS: install completed.

[19:21] [root@risadinha dkms]# ls -la /lib/modules/5.0.0-23-lowlatency/updates/dkms/hid*
-rw-r--r-- 1 root root    60613 dez 15 19:20 hid-logitech.ko
```

Patola, 2019-12-15:
Also, it doesn't install anything under /lib/modules/$(uname -r)/extra as the README says.

Patola, 2019-12-15:
And oh, sorry, Ubuntu 18.04 with latest upgrades, kernel-headers and all the right stuff. If you need any further details just ask.

berarma, 2019-12-16:
There are two installation methods. The module installs in the `extra` directory when using the manual install and it's called `hid-logitech-new`.

When using DKMS to manage the module, it gets installed as `hid-logitech` to make it easier to use. You can choose between the in-kernel and new-lg4ff module using DMKS install/uninstall commands. Everything is taken care of by DKMS.

I think it's correctly explained in the README but it may require a second read to understand the sparse information.

closed 2019-12-25 by berarma