berarma/new-lg4ff #19: Version 0.2 not working

writequit, 2019-12-26:
Hi Bernat,

Firstly thanks for taking the trouble to create a working driver for us,  I've been using version 0.1 in pcars2, ATS and Mudrunner and it's been working well. 

I updated to version 0.2 a short while ago using dkms, rebooted and connected the wheel and get an oops https://pastebin.com/aY4y5n31

Reverted to 0.1 and working again.

Regards,
Geoff

berarma, 2019-12-26:
This is the same as #20. Something is causing the wheel to disconnect and it triggers an error.

Thanks for reporting.

berarma, 2019-12-26:
It's fixed in the master branch, please test it and confirm it works.

writequit, 2019-12-26:
I confirm it's working without issue now, loaded the new module, no oops, couple of laps around the Nordschleife, all good :)

Thank you!

berarma, 2019-12-27:
Thanks for reporting.

closed 2019-12-27 by berarma