berarma/new-lg4ff #20: Version 0.2 error with G27 wheel

alexbrrsclnt, 2019-12-26:
Hi!

I have been using version 0.1 without any issue for a while. Today updated to 0.2 and I see this log when I connect the wheel to the usb port.

https://pastebin.com/qLLKEfet

Tell me if I can test any change

jotomo, 2019-12-26:
I'm also using a G27, but I don't get the error. Did you (and if not could you try) rebooting? Initially when plugging in, the green LED on the shifter only shortly switched on, so I rebooted to get into a clean state (didn't check the kernel output, but suspected a fault cable).

jotomo, 2019-12-26:
Okay, now I see it too when I unplug the wheel, the _deinit_ routine trips up, requiring a restart since _rmmod_ hangs removing the module.

berarma, 2019-12-26:
Nice find @jotomo. For @alexbrrsclnt, the wheel is disconnecting when loading the module. I don't know the reason, maybe unrelated to the driver, like a faulty cable.

I'll try to fix the error when disconnecting the cable. Hopefully this will fix the other problem too.

berarma, 2019-12-26:
For me, it's failing when connecting the wheel after the module is loaded. If that's what happens to you, a temporary workaround would be having the wheel already connected when booting.

berarma, 2019-12-26:
I've finally uncovered the real issue and it's fixed in master. Please, test it and confirm is fixed.

jotomo, 2019-12-26:
Tested unloading and loading and plugging/unplugging a few times. No more oops, wheel still working in-game :+1:

berarma, 2019-12-27:
Thanks for your reports. New release up.

closed 2019-12-27 by berarma