berarma/new-lg4ff #23: G29 red knob button presses

IOBYTE, 2020-02-05:
I have a G29 and I am not getting reliable button presses from the red knob on the wheel in speed dreams and jstest-gtk.  I do get reliable button presses (one press per click) in oversteer.

I'm not sure the driver is installed properly:
```
[    1.718286] usb 1-1.4: New USB device found, idVendor=046d, idProduct=c24f
[    1.718288] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.718289] usb 1-1.4: Product: G29 Driving Force Racing Wheel
[    1.718290] usb 1-1.4: Manufacturer: Logitech
[    1.722014] hidraw: raw HID events driver (C) Jiri Kosina
[    1.725198] usbcore: registered new interface driver usbhid
[    1.725198] usbhid: USB HID core driver
[    1.727177] hid_logitech_new: loading out-of-tree module taints kernel.
[    1.727178] hid_logitech_new: loading out-of-tree module taints kernel.
[    1.727214] hid_logitech_new: module verification failed: signature and/or required key missing - tainting kernel
[    1.727529] input: Logitech G29 Driving Force Racing Wheel as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:046D:C24F.0001/input/input5
[    1.727587] logitech 0003:046D:C24F.0001: input,hidraw0: USB HID v1.11 Joystick [Logitech G29 Driving Force Racing Wheel] on usb-0000:00:1a.0-1.4/input0
[    1.727666] logitech 0003:046D:C24F.0001: Force feedback support for Logitech Gaming Wheels (0.2b)
[    1.727667] logitech 0003:046D:C24F.0001: Hires timer: period = 2 ms

```
I'm using the latest (Feb 4) driver version with Ubuntu 18.04.

berarma, 2020-02-05:
I don't see anything out of the ordinary. Have you tried with the in-kernel module? I guess it will be exactly the same.

The red knob produces very short pulses.Oversteer reads all events from the wheel and always displays a button press for at least 100ms. That's the reason it produces consistent visual feedback for the red knob.

I'm not sure what jstest-gtk or Speed Dreams are doing but they might not be reading all events. They could be polling the device at a periodic interval.That could be a reason they're loosing button presses. In addition, jstest-gtk doesn't make the pulses last for longer in it interface so maybe the the window is not updating fast enough, or your eye is not fast enough to see the button press.

closed 2020-02-05 by berarma

IOBYTE, 2020-02-06:
I ran jstest --events and the events are there. jstest-gtk flashes sometimes so the button press time is too short to see. Speed Dreams rarely sees the changes (maybe 1 in 10) so it must be polling rather than collecting events.  I'll bring the problem up with them.  Thanks.

berarma, 2020-02-06:
Good. They can reach me if they need to.