berarma/new-lg4ff #24: Manual compiling issue

schabrackes, 2020-02-09:
Sorry but maybe I'm just too stupid, I get an error when compiling the code:

`
micha@gamertux ~/Desktop/new-lg4ff-0.3 $ LC_MESSAGES=C make
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/src/linux-4.19.97-gentoo'
  CC [M]  /home/micha/Desktop/new-lg4ff-0.3/hid-lgff.o
/home/micha/Desktop/new-lg4ff-0.3/hid-lgff.c:116:5: error: redefinition of ‘lgff_init’
  116 | int lgff_init(struct hid_device* hid)
      |     ^~~~~~~~~
In file included from /home/micha/Desktop/new-lg4ff-0.3/hid-lgff.c:23:
/home/micha/Desktop/new-lg4ff-0.3/hid-lg.h:13:19: note: previous definition of ‘lgff_init’ was here
   13 | static inline int lgff_init(struct hid_device *hdev) { return -1; }
      |                   ^~~~~~~~~
make[2]: *** [scripts/Makefile.build:304: /home/micha/Desktop/new-lg4ff-0.3/hid-lgff.o] Error 1
make[1]: *** [Makefile:1522: _module_/home/micha/Desktop/new-lg4ff-0.3] Error 2
make[1]: Leaving directory '/usr/src/linux-4.19.97-gentoo'
make: *** [Makefile:5: default] Error 2
`
I tried to compile it with 5.4 kernel sources but got the same result. Not sure what I#m doing wrong here. I've no experience in compiling code manually. Thanks!

berarma, 2020-02-10:
It seems you have a kernel configuration that disables some parts of the hid-logitech driver. new-lg4ff wasn't handling this case well. I've uploaded a fix. Please, can you try with the latest code in the master branch?

schabrackes, 2020-02-10:
> It seems you have a kernel configuration that disables some parts of the hid-logitech driver. new-lg4ff wasn't handling this case well. I've uploaded a fix. Please, can you try with the latest code in the master branch?

It's working now! Thanks! 
I tested ETS2 and Dirt Rally so far - feeling great!

berarma, 2020-02-10:
Thanks for the confirmation.

closed 2020-02-10 by berarma

ManoaNosea, 2020-03-02:
I have similar problem with a DFGT, I added led support, hid-logitech to the kernel:
```
manoa:/mnt/build/ffb/ffb2/new-lg4ff$ make -j8
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory `/mnt/build/linux-5.5.5'
  AR      /mnt/build/ffb/ffb2/new-lg4ff/built-in.a
  CC [M]  /mnt/build/ffb/ffb2/new-lg4ff/hid-lg.o
  CC [M]  /mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.o
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_init':
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2229:41: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  error = device_create_file(&hid->dev, &dev_attr_peak_ffb_level);
                                         ^~~~~~~~~~~~~~~~~~~~~~~
                                         dev_attr_damper_level
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2229:41: note: each undeclared identifier is reported only once for each function it appears in
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2245:41: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  error = device_create_file(&hid->dev, &dev_attr_ffb_leds);
                                         ^~~~~~~~~~~~~~~~~
                                         dev_attr_real_id
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_deinit':
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2328:33: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  device_remove_file(&hid->dev, &dev_attr_peak_ffb_level);
                                 ^~~~~~~~~~~~~~~~~~~~~~~
                                 dev_attr_damper_level
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2332:33: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  device_remove_file(&hid->dev, &dev_attr_ffb_leds);
                                 ^~~~~~~~~~~~~~~~~
                                 dev_attr_real_id
make[3]: *** [/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/mnt/build/ffb/ffb2/new-lg4ff] Error 2
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory `/mnt/build/linux-5.5.5'
make: *** [default] Error 2
```

berarma, 2020-03-02:
Please open a new issue and explain the steps you took to build.