berarma/new-lg4ff #25: Installing with DKMS not work

leotada, 2020-02-09:
I tried to install using DKMS on Fedora 31 (5.4.17-200.fc31.x86_64) and DKMS status show this:

```
$ sudo dkms status
new-lg4ff, 0.3, 5.4.17-200.fc31.x86_64, x86_64: installed
```

but on dmesg command do not show any change, and only work like the old kernel driver.
Using the manual installations works and show the 0.3 version on dmesg.

jotomo, 2020-02-09:
Did you `rmmod hid-logitech` the old driver and loaded the new one (`modprobe hid-logitech`)? Check with `lsmod`, which shows the new drivers as `hid_logitech_new`, whereas the old one doesn't have the `new` suffix.

leotada, 2020-02-09:
I don't know much about kernel modules, I used only the command on the readme:
sudo dkms install /usr/src/new-lg4ff
I hadn't installed any version yet. This command will build and load the module, right?

jotomo, 2020-02-09:
Yes, `dkms install` builds and installs the module but doesn't load or unload the module.
Alternatively, cou can also just restart afterwards to have the new module loaded.

leotada, 2020-02-09:
I restarted the computer thinking this, but didn't load the new driver module automatically, it didn't work, for some reason.

berarma, 2020-02-09:
> I restarted the computer thinking this, but didn't load the new driver module automatically, it didn't work, for some reason.

This will only work when installing the module using DKMS. It should install the driver and update the iniramfs image with the new module. Pasting the output of the dkms install command could help determine what's happening.

leotada, 2020-02-10:
> $ sudo dkms install /usr/src/new-lg4ff
> 
> Creating symlink /var/lib/dkms/new-lg4ff/0.3/source ->
>                  /usr/src/new-lg4ff-0.3
> 
> DKMS: add completed.
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> make -j16 KERNELRELEASE=5.4.17-200.fc31.x86_64 KVERSION=5.4.17-200.fc31.x86_64...
> cleaning build area...
> 
> DKMS: build completed.
> 
> hid-logitech.ko.xz:
> Running module version sanity check.
>  - Original module
>    - No original module exists within this kernel
>  - Installation
>    - Installing to /lib/modules/5.4.17-200.fc31.x86_64/extra/
> Adding any weak-modules
> 
> depmod....
> 
> Backing up initramfs-5.4.17-200.fc31.x86_64.img to /boot/initramfs-5.4.17-200.fc31.x86_64.img.old-dkms
> Making new initramfs-5.4.17-200.fc31.x86_64.img
> (If next boot fails, revert to initramfs-5.4.17-200.fc31.x86_64.img.old-dkms image)
> dracut.........
> 
> DKMS: install completed.

berarma, 2020-02-10:
That looks good. Could you attach the output of the dmesg command after rebooting?

leotada, 2020-02-10:
> That looks good. Could you attach the output of the dmesg command after rebooting?

$ sudo dmesg | grep logitech
[sudo] senha para leonardo: 
[  159.603595] logitech 0003:046D:C294.0006: fixing up Logitech Driving Force report descriptor
[  159.604004] logitech 0003:046D:C294.0006: input,hidraw5: USB HID v1.00 Joystick [G27 Racing Wheel] on usb-0000:01:00.0-4/input0
[  160.415189] logitech 0003:046D:C29B.0007: input,hidraw5: USB HID v1.11 Joystick [G27 Racing Wheel] on usb-0000:01:00.0-4/input0
[  160.415293] logitech 0003:046D:C29B.0007: Force feedback support for Logitech Gaming Wheels

berarma, 2020-02-10:
Could it be the complete output of the dmesg command after a reboot? We're trying to find something that has gone wrong and I need more information about your system. Are you sure you're running the same kernel that has the newly installed kernel?

leotada, 2020-02-10:
I only restarted the computer and booted on the same version of kernel.
[dmesg.txt](https://github.com/berarma/new-lg4ff/files/4182473/dmesg.txt)
[lsmod.txt](https://github.com/berarma/new-lg4ff/files/4182475/lsmod.txt)

berarma, 2020-02-11:
The DKMS installation process seems to go well but the module is not found at reboot. We aren't doing anything weird on purpose. This module should load like any other DKMS module. Could you ask in some Fedora support forum? They may have some ideas as to why it doesn't work.

bam085, 2020-02-12:
I'm having the same issue on Fedora 31. I haven't spent much time 
figuring out why because I was so eager to try this so I followed the 
steps under manual method and that works.

berarma, 2020-02-12:
> I'm having the same issue on Fedora 31. I haven't spent much time figuring out why because I was so eager to try this so I followed the steps under manual method and that works.

Thanks for reporting. I've never used Fedora and I'm no expert on DKMS. Any external help is welcomed. Some Fedora users may know better than me.

I've done a search for "fedora 31 dkms" and I got this result: https://github.com/zfsonlinux/zfs/issues/9868

That leads me to: https://bugzilla.redhat.com/show_bug.cgi?id=1753044

I'm not sure that it's the same exact issue but seems very related. Installing the grubby package might be worth a try.

jp7677, 2020-02-25:
@berarma I'm running Fedora 31, thanks a lot for the link to that issue. That issue is now resolved with a new dkms release which has been pushed to Fedora 31. With that update the dkms procedure for installing lg4ff-new is indeed working now.

berarma, 2020-02-25:
@jp7677 Thanks for the confirmation!

closed 2020-02-25 by berarma