berarma/new-lg4ff #26: vibration in wheel when slowly turning against a const force

IOBYTE, 2020-03-03:
I'm having a problem with a vibration in the wheel when slowly turning the wheel against a constant force. I noticed this in speed dreams and vdrift in certain corners on certain tracks.  I'm able to reproduce the problem with` fftest_buffer_overrun` (https://github.com/Eliasvan/Linux-Force-Feedback/tree/master/tools/RateLimiting).

I had to increase the device buffer length in fftest_buffer_overrun.c to get fftest_buffer_overrun to work.
```
diff --git a/tools/RateLimiting/fftest_buffer_overrun.c b/tools/RateLimiting/fftest_buffer_overrun.c
index 25a46ef..00c8784 100644
--- a/tools/RateLimiting/fftest_buffer_overrun.c
+++ b/tools/RateLimiting/fftest_buffer_overrun.c
@@ -33,7 +33,7 @@ enum {

 void print_help(char *device_file_name, unsigned long update_period, unsigned long total_time)
 {
-   printf("Usage: %s /dev/input/eventXX [updatePeriodMicros (default=%d) [totalTimeMicros (default=%d)]]\n",
+   printf("Usage: %s /dev/input/eventXX [updatePeriodMicros (default=%lu) [totalTimeMicros (default=%lu)]]\n",
            device_file_name, update_period, total_time);
    printf("Tests the force feedback driver\n");
    exit(1);
@@ -45,7 +45,7 @@ int main(int argc, char **argv)
    struct input_event play, stop;
    unsigned long update_period, total_time;
    int fd;
-   char device_file_name[64];
+   char device_file_name[512];
    int i;

    printf("Force feedback test program.\n");
@@ -63,7 +63,7 @@ int main(int argc, char **argv)
            print_help(argv[0], update_period, total_time);
        switch (i) {
        case 1:
-           strncpy(device_file_name, argv[i], 64);
+           strncpy(device_file_name, argv[i], sizeof(device_file_name));
            break;
        case 2:
            update_period = atol(argv[i]);

```
```
~/Linux-Force-Feedback/tools/RateLimiting$ ./fftest_buffer_overrun /dev/input/by-id/usb-Logitech_G29_Driving_Force_Racing_Wheel-event-joystick 1000 10000000
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/by-id/usb-Logitech_G29_Driving_Force_Racing_Wheel-event-joystick opened
Now Playing CONSTANT/RUMBLE effect with almost unnoticable magnitude...
Now Playing CONSTANT/RUMBLE effect with large magnitude...
The CONSTANT/RUMBLE effect has been stopped.
```
If feel the vibration when slowly turning the wheel in the direction of the force.

I started having this problem when I installed this driver.  I don't have this problem when playing other racing games on Windows 7.

I would like to see if others can reproduce the problem. I would like to know if my wheel is going bad or that's just the way the g29 works.

parkerlreed, 2020-03-03:
```
[parker@wolfcola Linux-Force-Feedback]$ patch -p1 < pa
patching file tools/RateLimiting/fftest_buffer_overrun.c
Hunk #1 FAILED at 33.
Hunk #2 FAILED at 45.
Hunk #3 FAILED at 63.
3 out of 3 hunks FAILED -- saving rejects to file tools/RateLimiting/fftest_buffer_overrun.c.rej
```

berarma, 2020-03-03:
> I started having this problem when I installed this driver. I don't have this problem when playing other racing games on Windows 7.

You should compare this driver with the in-kernel driver module. Comparisons with Windows can be useful if you can reproduce exactly the same conditions in both systems.

> I would like to see if others can reproduce the problem. I would like to know if my wheel is going bad or that's just the way the g29 works.

The Logitech wheels may forward vibrations under certain circumstances because of its internal mechanics. That's why you should compare under the same conditions.

IOBYTE, 2020-03-04:
I think I'm running the in-kernel now:
```
$ lsmod | grep logitech
hid_logitech           32768  0
ff_memless             16384  1 hid_logitech
hid                   114688  4 hid_logitech,usbhid,hid_microsoft,hid_generic
```
```
$ sudo modinfo hid-logitech.ko
filename:       /lib/modules/4.15.0-88-generic/kernel/drivers/hid/hid-logitech.ko
license:        GPL
srcversion:     A39F535BC25752ACEA770DD
alias:          hid:b0003g*v0000046Dp0000C623
alias:          hid:b0003g*v0000046Dp0000C626
alias:          hid:b0003g*v0000046Dp0000C287
alias:          hid:b0003g*v0000046Dp0000C218
alias:          hid:b0003g*v0000046Dp0000C293
alias:          hid:b0003g*v0000046Dp0000C29C
alias:          hid:b0003g*v0000046Dp0000C298
alias:          hid:b0003g*v0000046Dp0000C29B
alias:          hid:b0003g*v0000046Dp0000C29A
alias:          hid:b0003g*v0000046Dp0000C299
alias:          hid:b0003g*v0000046Dp0000CA04
alias:          hid:b0003g*v0000046Dp0000CA03
alias:          hid:b0003g*v0000046Dp0000C295
alias:          hid:b0003g*v0000046Dp0000C286
alias:          hid:b0003g*v0000046Dp0000C283
alias:          hid:b0003g*v0000046Dp0000C24F
alias:          hid:b0003g*v0000046Dp0000C219
alias:          hid:b0003g*v0000046Dp0000C211
alias:          hid:b0003g*v0000046Dp0000C20A
alias:          hid:b0003g*v0000046Dp0000C294
alias:          hid:b0003g*v0000046Dp0000C216
alias:          hid:b0003g*v0000046Dp0000C215
alias:          hid:b0003g*v0000046Dp0000C512
alias:          hid:b0003g*v0000046Dp0000C30A
alias:          hid:b0003g*v0000046Dp0000C71F
alias:          hid:b0003g*v0000046Dp0000C714
alias:          hid:b0003g*v0000046Dp0000C704
alias:          hid:b0003g*v0000046Dp0000C101
alias:          hid:b0003g*v0000046Dp0000C517
alias:          hid:b0003g*v0000046Dp0000C50C
alias:          hid:b0003g*v0000046Dp0000C513
depends:        hid,ff-memless
retpoline:      Y
intree:         Y
name:           hid_logitech
vermagic:       4.15.0-88-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4
parm:           lg4ff_no_autoswitch:Do not switch multimode wheels to their native mode automatically (int)
```
The in-kernel driver behaves the same way.  I tried various update rates with no change.

IOBYTE, 2020-03-05:
I rebooted with the in-kernel driver and the vibration is gone.  Now when I turn the wheel with a constant force there is a notchy feeling.  The 2 drivers do behave differently.

berarma, 2020-03-05:
@parkerlreed, I could apply the patch with the command `patch -n -p1'.

@IOBYTE, I've tried fftest_buffer_overrun and there's only a slight variable vibration with both the old and the new driver. It's exactly the same. Could you make a video testing with both drivers?

berarma, 2020-03-06:
I didn't try to reboot with the in-kernel driver. I'll try it when I get home.

IOBYTE, 2020-03-06:
I had a hard time removing your driver.  I finally ended up with:
```
sudo dkms remove new-lg4ff/0.3 --all
sudo update-initramfs -u -k all
```
and rebooting.

I'm not sure about a video since the the vibration is in the force feedback of the wheel.  You can hear the wheel making a noise but it is faint.

The way I reproduce the problem is to run `fftest_buffer_overrun` and hold the wheel tight. When the constant force starts I slowly turn the wheel counter-clockwise.

With the original driver the force feedback feels like it gets stronger and then returns to normal about once a second.  This makes the wheel feel like it has notches. 

With the new driver force feedback feels like it increases and returns to normal maybe 30 times a second which makes the wheel feel like it's vibrating.

You can feel the effect in both drivers when driving in the games but it is much more noticeable with the new driver.

berarma, 2020-03-06:
YES! You were right! Thanks for your patience.

It's the friction effect that, when downloaded to the wheel, causes weird effects even if its parameters are set to zero when not in use

I'll have to change the code so that it only downloads the effect when it's being effectively used and removed when not.

> I had a hard time removing your driver. I finally ended up with:
> 
> ```
> sudo dkms remove new-lg4ff/0.3 --all
> sudo update-initramfs -u -k all
> ```
> 
> and rebooting.

`dkms remove` and rebooting should be enough unless you're using an old version of the driver. Please, open a new issue if you think it's not working correctly.

berarma, 2020-03-07:
I've updated master with a fix. Can you try it?

Now the friction effect won't be used unles the game uses it. Also, it can be completely disabled from Oversteer if we don't like how it works on certain games that use it.

IOBYTE, 2020-03-07:
Thanks. The vibration is gone and turning is smooth now.

In speed dreams I don't feel the curbs and rumble strips with the default settings.  When I increase the force feedback and disable the smoothing I can feel the curbs and rumble strips but they don't feel as realistic as rfactor 2, race room or project cars 2 on Windows. I don't remember how it worked with the old driver. I assume the problem is in the game so I will try with the old driver when I get a chance to make sure it is not the new driver. I'll also use ffbwrap again to see what the game is trying to do.

leillo1975, 2020-03-07:
Perhapas @madbad could help with the Speed Dreams FFB

berarma, 2020-03-08:
> In speed dreams I don't feel the curbs and rumble strips with the default settings. When I increase the force feedback and disable the smoothing I can feel the curbs and rumble strips but they don't feel as realistic as rfactor 2, race room or project cars 2 on Windows.

You're comparing different games on different operating systems. There's no point in that. Try to do the comparation using the old and new driver with the same game in the same operating system. Speed Dreams uses only constant force so it should be the same.

IOBYTE, 2020-03-08:
Speed Dreams also uses periodic sine:
```
000000000000 # DEVICE_NAME=Logitech G29 Driving Force Racing Wheel, UPDATE_FIX=0,DIRECTION_FIX=0, FEATURES_HACK=0,FORCE_INVERSION=0, IGNORE_SET_GAIN=0, OFFSET_FIX=0
000000501735 > QUERY # Query force feedback features.
000000501781 < 16,  Constant Periodic ( Square Triangle Sine Saw up Saw down ) Ramp Spring Friction Damper Rumble Gain Autocenter
000000789786 > QUERY # Query force feedback features.
000000789819 < 16,  Constant Periodic ( Square Triangle Sine Saw up Saw down ) Ramp Spring Friction Damper Rumble Gain Autocenter
000000789857 > QUERY # Query force feedback features.
000000789871 < 16,  Constant Periodic ( Square Triangle Sine Saw up Saw down ) Ramp Spring Friction Damper Rumble Gain Autocenter
000000789891 > QUERY # Query force feedback features.
000000789920 < 16,  Constant Periodic ( Square Triangle Sine Saw up Saw down ) Ramp Spring Friction Damper Rumble Gain Autocenter
000000789928 > SLOTS # Get maximum number of simultaneous effects in memory.
000000789936 < 0, effects: 16
000000789955 > UPLOAD id:-1 dir:16384 length:1000 delay:0 type:CONSTANT level:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000000789966 < 0 id:0
000000789977 > PLAY 0 2147483647
000000789991 < 24
000000790004 > UPLOAD id:0 dir:16384 length:1000 delay:0 type:CONSTANT level:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000000790012 < 0 id:0
000000790031 > UPLOAD id:-1 dir:16384 length:5000 delay:0 type:PERIODIC waveform:SINE period:1000 magnitude:16384 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000000790040 < 0 id:1
000000790055 > UPLOAD id:1 dir:16384 length:100 delay:0 type:PERIODIC waveform:SINE period:1000 magnitude:16383 offset:0 phase:0 attack_length:0 attack_level:0 fade_length:0 fade_level:0
000000790063 < 0 id:1
000000790071 > PLAY 1 1
000000790079 < 24
000017181759 > REMOVE 0 # Remove effect from memory.
000017181801 < 0
000017181813 > REMOVE 1 # Remove effect from memory.
000017181822 < 0
```

berarma, 2020-03-09:
I was told it used only constant force. I haven't tested it very much. @leillo1975, @madbad?

Still, you should be more specific about what's wrong related to this project, not the game. I don't know why you think there's a problem in the FFB.

IOBYTE, 2020-03-09:
I'm using windows behavior just to rule out my wheel as the problem.  I'm using Speed Dreams behavior to try to rule out your driver. I wanted to make sure the two effects are playing well together. I believe the remaining problems I'm having now are game related.  ffbmeter is indicating clipping when turning over curbs.

This issue can be closed because the reported problem is now fixed.

closed 2020-03-09 by IOBYTE