berarma/new-lg4ff #27: compiling problem with kerenal 5.5.5

ManoaNosea, 2020-03-03:
I have similar problem with a DFGT, I added led support, hid-logitech to the kernel:
```
manoa:/mnt/build/ffb/ffb2/new-lg4ff$ make -j8
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory `/mnt/build/linux-5.5.5'
  AR      /mnt/build/ffb/ffb2/new-lg4ff/built-in.a
  CC [M]  /mnt/build/ffb/ffb2/new-lg4ff/hid-lg.o
  CC [M]  /mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.o
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_init':
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2229:41: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  error = device_create_file(&hid->dev, &dev_attr_peak_ffb_level);
                                         ^~~~~~~~~~~~~~~~~~~~~~~
                                         dev_attr_damper_level
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2229:41: note: each undeclared identifier is reported only once for each function it appears in
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2245:41: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  error = device_create_file(&hid->dev, &dev_attr_ffb_leds);
                                         ^~~~~~~~~~~~~~~~~
                                         dev_attr_real_id
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_deinit':
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2328:33: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  device_remove_file(&hid->dev, &dev_attr_peak_ffb_level);
                                 ^~~~~~~~~~~~~~~~~~~~~~~
                                 dev_attr_damper_level
/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.c:2332:33: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  device_remove_file(&hid->dev, &dev_attr_ffb_leds);
                                 ^~~~~~~~~~~~~~~~~
                                 dev_attr_real_id
make[3]: *** [/mnt/build/ffb/ffb2/new-lg4ff/hid-lg4ff.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/mnt/build/ffb/ffb2/new-lg4ff] Error 2
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory `/mnt/build/linux-5.5.5'
make: *** [default] Error 2
```
@berarma
Owner
berarma commented Mar 2, 2020
Please open a new issue and explain the steps you took to build.

no special steps, just runned "make" with gcc 8.3.1 :)
at first I tryed to build on a kerenal without LED support and without hid-logitech
it gived mutch more build errors (redefinition errors)
the module is working after I did the modifications on this codes and ffb is working :)
I decide to send this report anyway becuase I guess you whant to fix any problems :)

berarma, 2020-03-03:
OK. Your build errors are caused by the leds support being disabled. I guess it built correctly after you enabled leds support in the kernel. Is that correct?

Thanks for the report.

ManoaNosea, 2020-03-03:
no, this specific error happen when leds and hid-logitech was enable in the kerenal :x
the error with them disabled was mutch bigger, mostly redefinition errors

ManoaNosea, 2020-03-03:
it funny that redefinition errors happening when hid-logitech don't exist at all, or could it be because the hid-generic defined them instead ?

ManoaNosea, 2020-03-03:
to make compile I had to remove this 2 codes from source

ManoaNosea, 2020-03-03:
...I think you should give linux torvalds the codes :) it realy grate driver :)

berarma, 2020-03-03:
Could you paste here the redefinition errors?

Also, which codes did you remove?

Sorry, I'm having a hard time understanding you.

berarma, 2020-03-03:
> ...I think you should give linux torvalds the codes :) it realy grate driver :)

Thanks! Let's first fix all the bugs.

ManoaNosea, 2020-03-04:
```
manoa:/mnt/build/ffb/ffb3/new-lg4ff$ make -j8
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory `/mnt/build/linux-5.5.5'
  CC [M]  /mnt/build/ffb/ffb3/new-lg4ff/hid-lg.o
  CC [M]  /mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.o
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:1037:5: error: redefinition of 'lg4ff_adjust_input_event'
 int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.h:15:19: note: previous definition of 'lg4ff_adjust_input_event' was here
 static inline int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
                   ^~~~~~~~~~~~~~~~~~~~~~~~
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:1063:5: error: redefinition of 'lg4ff_raw_event'
 int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
     ^~~~~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.h:17:19: note: previous definition of 'lg4ff_raw_event' was here
 static inline int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
                   ^~~~~~~~~~~~~~~
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_alternate_modes_store':
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:1539:60: error: 'lg4ff_no_autoswitch' undeclared (first use in this function); did you mean 'lg4ff_compat_mode_switch'?
  if (target_product_id == USB_DEVICE_ID_LOGITECH_WHEEL && !lg4ff_no_autoswitch) {
                                                            ^~~~~~~~~~~~~~~~~~~
                                                            lg4ff_compat_mode_switch
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:1539:60: note: each undeclared identifier is reported only once for each function it appears in
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_handle_multimode_wheel':
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2064:7: error: 'lg4ff_no_autoswitch' undeclared (first use in this function); did you mean 'lg4ff_compat_mode_switch'?
      !lg4ff_no_autoswitch) {
       ^~~~~~~~~~~~~~~~~~~
       lg4ff_compat_mode_switch
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: At top level:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2089:5: error: redefinition of 'lg4ff_init'
 int lg4ff_init(struct hid_device *hid)
     ^~~~~~~~~~
In file included from /mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.h:19:19: note: previous definition of 'lg4ff_init' was here
 static inline int lg4ff_init(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_init':
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2229:41: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  error = device_create_file(&hid->dev, &dev_attr_peak_ffb_level);
                                         ^~~~~~~~~~~~~~~~~~~~~~~
                                         dev_attr_damper_level
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2245:41: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  error = device_create_file(&hid->dev, &dev_attr_ffb_leds);
                                         ^~~~~~~~~~~~~~~~~
                                         dev_attr_real_id
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: At top level:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2298:5: error: redefinition of 'lg4ff_deinit'
 int lg4ff_deinit(struct hid_device *hid)
     ^~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.h:20:19: note: previous definition of 'lg4ff_deinit' was here
 static inline int lg4ff_deinit(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~~~
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_deinit':
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2328:33: error: 'dev_attr_peak_ffb_level' undeclared (first use in this function); did you mean 'dev_attr_damper_level'?
  device_remove_file(&hid->dev, &dev_attr_peak_ffb_level);
                                 ^~~~~~~~~~~~~~~~~~~~~~~
                                 dev_attr_damper_level
/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.c:2332:33: error: 'dev_attr_ffb_leds' undeclared (first use in this function); did you mean 'dev_attr_real_id'?
  device_remove_file(&hid->dev, &dev_attr_ffb_leds);
                                 ^~~~~~~~~~~~~~~~~
                                 dev_attr_real_id
make[3]: *** [/mnt/build/ffb/ffb3/new-lg4ff/hid-lg4ff.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/mnt/build/ffb/ffb3/new-lg4ff] Error 2
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory `/mnt/build/linux-5.5.5'
make: *** [default] Error 2
```
this what happen when leds disabled (which also disable the hid-logitech option completely - it just doesn't exist in make xconfig)

ManoaNosea, 2020-03-04:
[the modifications](http://manoa.dream.org.il/Registry/removed-codes.patch)

berarma, 2020-03-07:
I've updated master with a fix for this isuue. Can you try it?

ManoaNosea, 2020-03-07:
still some problems:
```
manoa:/mnt/build/ffb/ffb4/new-lg4ff$ make -j8
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory `/mnt/build/linux-5.5.5'
  AR      /mnt/build/ffb/ffb4/new-lg4ff/built-in.a
  CC [M]  /mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.o
  CC [M]  /mnt/build/ffb/ffb4/new-lg4ff/hid-lg.o
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:1073:5: error: redefinition of 'lg4ff_adjust_input_event'
 int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.h:15:19: note: previous definition of 'lg4ff_adjust_input_event' was here
 static inline int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
                   ^~~~~~~~~~~~~~~~~~~~~~~~
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:1099:5: error: redefinition of 'lg4ff_raw_event'
 int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
     ^~~~~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.h:17:19: note: previous definition of 'lg4ff_raw_event' was here
 static inline int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
                   ^~~~~~~~~~~~~~~
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_alternate_modes_store':
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:1575:60: error: 'lg4ff_no_autoswitch' undeclared (first use in this function); did you mean 'lg4ff_compat_mode_switch'?
  if (target_product_id == USB_DEVICE_ID_LOGITECH_WHEEL && !lg4ff_no_autoswitch) {
                                                            ^~~~~~~~~~~~~~~~~~~
                                                            lg4ff_compat_mode_switch
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:1575:60: note: each undeclared identifier is reported only once for each function it appears in
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c: In function 'lg4ff_handle_multimode_wheel':
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:2100:7: error: 'lg4ff_no_autoswitch' undeclared (first use in this function); did you mean 'lg4ff_compat_mode_switch'?
      !lg4ff_no_autoswitch) {
       ^~~~~~~~~~~~~~~~~~~
       lg4ff_compat_mode_switch
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c: At top level:
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:2125:5: error: redefinition of 'lg4ff_init'
 int lg4ff_init(struct hid_device *hid)
     ^~~~~~~~~~
In file included from /mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.h:19:19: note: previous definition of 'lg4ff_init' was here
 static inline int lg4ff_init(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:2338:5: error: redefinition of 'lg4ff_deinit'
 int lg4ff_deinit(struct hid_device *hid)
     ^~~~~~~~~~~~
In file included from /mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.c:22:
/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.h:20:19: note: previous definition of 'lg4ff_deinit' was here
 static inline int lg4ff_deinit(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~~~
make[3]: *** [/mnt/build/ffb/ffb4/new-lg4ff/hid-lg4ff.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/mnt/build/ffb/ffb4/new-lg4ff] Error 2
make[1]: *** [sub-make] Error 2
make[1]: Leaving directory `/mnt/build/linux-5.5.5'
make: *** [default] Error 2
```

ManoaNosea, 2020-03-07:
thare is another small problem, I played grid today a few times and I notice the brakes turn on by themselfs sometimes, the gas also turn off by itself sometimes, it's random

ManoaNosea, 2020-03-07:
at first I thought it was a analog problem, but then it happen also when deth zone was 5%

berarma, 2020-03-07:
> still some problems:
>
> ```
> manoa:/mnt/build/ffb/ffb4/new-lg4ff$ make -j8
> make -C /lib/modules/`uname -r`/build M=$PWD
> make[1]: Entering directory `/mnt/build/linux-5.5.5'
>   AR [...]

You're seeing those errors because you've disabled Logitech Wheel FFB in the kernel config. It's the CONFIG_LOGIWHEEELS_FF config setting that is disabled. I could try to fix it but there's absolutely no point in doing that and trying to use this project. What are you trying to accomplish?

berarma, 2020-03-07:
> thare is another small problem, I played grid today a few times and I notice the brakes turn on by themselfs sometimes, the gas also turn off by itself sometimes, it's random

This is a separate issue. I'm going to close this issue as long as the build problems are solved. Please, open a new one after verifying it doesn't happen with the in-kernel driver module.

ManoaNosea, 2020-03-08:
```
Logitech devices (HID_LOGITECH)

CONFIG_HID_LOGITECH:

Support for Logitech devices that are not fully compliant with HID standard.

Symbol: HID_LOGITECH [=n]
Type : tristate
Prompt: Logitech devices
Location:
-> Device Drivers
-> HID support
-> HID bus support (HID [=y])
-> Special HID drivers
Defined at drivers/hid/Kconfig:525
Depends on: INPUT [=y] && HID [=y] && LEDS_CLASS [=n]
```
the problem is that wheel doesn't have any leds but the kerenal driver depends on leds support :x

ManoaNosea, 2020-03-08:
I don't understand what that should be like that ?
what if someone whant the ffb but not any of the leds ?
it's not your problem I understand, but then you can't fix dependent on led support with this :x becuase the kerenal don't give that in config :x

ManoaNosea, 2020-03-08:
to get to the CONFIG_LOGIWHEEELS_FF you have to enable CONFIG_HID_LOGITECH but that depends on LEDS_CLASS, that the problem :x

berarma, 2020-03-08:
> to get to the CONFIG_LOGIWHEEELS_FF you have to enable CONFIG_HID_LOGITECH but that depends on LEDS_CLASS, that the problem :x

That would be a kernel issue then. I don't know the reason for this dependency, it should be possible to enable the FFB but not the leds. I can't fix that in the driver. Can you open an issue in the Linux project?

ManoaNosea, 2020-03-08:
I will try, thank :)

ManoaNosea, 2020-03-08:
I modded the lines in drivers/hid/Kconfig:525
and the driver compile with no problems :)

ManoaNosea, 2020-03-08:
it looks like leds requested by a driver for logitech gaming keybords
drivers/hid/hid-lg-g15

berarma, 2020-03-08:
Yes, that `depends` declaration seems to be the problem. You should bring up this issue in the Linux bugtracker because it's out of the scope of this project.

The issue here is already fixed so I'll close this.

closed 2020-03-08 by berarma