berarma/new-lg4ff #28: wheel not centered after boot

IOBYTE, 2020-04-12:
At boot time the wheel is turned from stop to stop and then is centered but it is centered one turn off.  That means when I start a race the car makes a hard left turn and I have to turn it a full turn right to go straight.

The wheel is centered for straight ahead in windows.

berarma, 2020-04-13:
The device firmware does the calibration and the almost centering on its own. We would have to wait a few seconds (I don't know how many) for the calibration to end and then issue an autocentering effect for a few seconds again to hopefully center the wheel. I don't think the driver should be managing the wheel position in this way. The in-kernel driver doesn't and I don't think this should change.

[Oversteer](https://github.com/berarma/oversteer) can be used to center the wheel by running it at boot with command line parameters. It can also be run from the game launcher. Some games also solve it by autocentering the wheel at start or showing a virtual wheel on screen.

I have the feeling that implementing this in the driver may cause more issues than it solves. We may leave this issue open though for further thinking.

berarma, 2021-01-02:
I've finally implemented this feature in [Oversteer](/berarma/oversteer). I think it will be more useful there and it doesn't mess with the driver at the lowest level. You can run Oversteer at any moment, when booting and when starting an app to center the wheel.

Thanks for the idea.

closed 2021-01-02 by berarma