berarma/new-lg4ff #30: Force Feedback doesn't work with Project Cars 2

TobiPeterG, 2020-06-14:
Hello guys,

this driver is really fantastic, thanks for your great work!
However, I have a problem with Project Cars 2 as I don't have any Force Feedback effects, though, the racing wheel is recognized and I can use it.
Is that a known problem?
Have I done something wrong?

Thanks in advance!

berarma, 2020-06-14:
Thanks for your compliments.

It should work without problems. It's not even needed for PC2, it should work with the in-kernel driver too. Try with the latest Proton version, if it doesn't work you could try the latest 4.11 version too. Older Proton versions have FFB bugs.

closed 2020-06-14 by berarma

TobiPeterG, 2020-06-15:
> Thanks for your compliments.
> 
> It should work without problems. It's not even needed for PC2, it should work with the in-kernel driver too. Try with the latest Proton version, if it doesn't work you could try the latest 4.11 version too. Older Proton versions have FFB bugs.

Oh you're right, I'm using a custom proton version and that doesn't work.
But default Proton 5.0 works, thanks, sorry for the inconvenience.