berarma/new-lg4ff #34: No damping or friction with G29 in some games/programs

janibonnevier, 2020-11-10:
Everything works fine in Eurotruck Sim 2, F1 2017 and Dirt Rally 2.0 (as far as I can tell). 

In F1 2019 (Proton 5.13-1) there doesn't seem to be any damping or friction effects whatsoever (other effects do work). 

In Oversteer, adjusting the sliders for force feedback gain, and spring, damper and friction force levels has no effect.

In fftest i get a listing the following effects, but effects 2, 4 and 5 do not work. Interestingly the damper effect does work here.

```
Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... OK (id 1)
Uploading effect #2 (Spring) ... OK (id 2)
Uploading effect #3 (Damper) ... OK (id 3)
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 4)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 5)
```

Unless you have some ideas straight away, I'll be happy to do some more digging, if only I get some instructions on how to do so.

Thank you for your work, it is much appreciated.

berarma, 2020-11-10:
> In F1 2019 (Proton 5.13-1) there doesn't seem to be any damping or friction effects whatsoever (other effects do work).

I don't own this game. How do you test it? Maybe @leillo1975 can explain it.

> In Oversteer, adjusting the sliders for force feedback gain, and spring, damper and friction force levels has no effect.

These sliders doesn't play any effect, it just changes the global gain and individual conditional effects gains. What do you expect?

> In fftest I get a listing the following effects, but effects 2, 4 and 5 do not work. Interestingly the damper effect does work here.
>
> ```
> Setting master gain to 75% ... OK
> Uploading effect #0 [...]

Effect 2 is playing for me but is very light. You'll notice some resistance when moving the wheel away from the center. Increasing the spring gain in oversteer will make it stronger.

I have to test effects 4 and 5 in more detail.

leillo1975, 2020-11-10:
For F1 2019 try to use Proton 5.0-10 or Proton 5.9-GE:

https://github.com/GloriousEggroll/proton-ge-custom

janibonnevier, 2020-11-10:
@berarma I was expecting the feel of the wheel to change right then and there, like it does with autocenter, but I had just misunderstood how it's supposed to work. You are right, fftest effect 2 works when I increase the spring level in Oversteer.

I guess this is simply an issue specific to F1 2019 then. @leillo1975 Are you implying that you are getting some kind of resistance when turning the wheel in F1 2019? Mine is just turning completely freely. Were your issues with Proton 5.13 related to FFB or something else? I'll experiment some more with different Proton versions, but I've never noticed any difference with FFB in the past.

leillo1975, 2020-11-11:
With Proton 5.0 and Proton-GE, I have resistance on wheel when I turn to the left and right, and everything seems to be ok. I reported this on the Proton Issue:
https://github.com/ValveSoftware/Proton/issues/2881#issuecomment-716139831
Can you leave a report attesting to what I have said in case it works for you? Let's see if they keep an eye on that regression.

berarma, 2020-11-11:
@janibonnevier: Expanding on `fftest`, it's very buggy. Maybe because it was created when only joysticks were supported, it doesn't apply very well to wheels. Some effects parameters aren't initialized thus they're set to random values and create an effect with near zero strength. I think I should add a warning in the README. I sent a patch for another bug in `ffcstress` some time ago but no one noticed it. It seems the project is unmaintained.

About F1 2019, we all three have the same wheel, I remember @leillo1975 told me the game had very different, and sometimes weird, FFB depending on the Proton version. It's the only game with issues, and since FFB code isn't changing that often I guess it might be something else in Proton/Wine.

janibonnevier, 2020-11-11:
@berarma @leillo1975 Thanks. I'll have a look at that Proton issue. Someone there is describing very similar issues to what I'm having.

closed 2021-04-12 by berarma