berarma/new-lg4ff #35: Upstreaming the driver

RyuzakiKK, 2020-12-08:
Is there any plan for upstreaming this module driver to expand/replace the current in-tree `hid-logitech`?

berarma, 2020-12-08:
That's what I'd like to do. I've been planning it in my head but I think there's some work to do and I'm not sure how well will it be received. I just need some motivation and a bit of time.

I think I should break the changes into several patches for each feature, there would be a big one for the effects implementation and other smaller patches for the secondary features. And for every patch I work on, send it upstream and try to get it merged.

Any advice and guiding from someone knowledgeable would help at this point.

Elizafox, 2021-03-29:
> That's what I'd like to do. I've been planning it in my head but I think there's some work to do and I'm not sure how well will it be received. I just need some motivation and a bit of time.
>
> I [...]

Hi :smiley: 

I have contributed to Linux before, although only a minor patch (adding a device ID to a serial device, many moons ago). But! It isn't too difficult and don't get too intimidated.

It looks like the HID device maintainer list is at linux-input@vger.kernel.org, as found in the `MAINTAINERS` file:

```
HID CORE LAYER
M:	Jiri Kosina <jikos@kernel.org>
M:	Benjamin Tissoires <benjamin.tissoires@redhat.com>
L:	linux-input@vger.kernel.org
S:	Maintained
T:	git git://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git
F:	drivers/hid/
F:	include/linux/hid*
F:	include/uapi/linux/hid*
```

It's best to touch base with them first and get feedback on your changes and how they want to go about this.

Make sure all your code follows [the kernel coding style](https://www.kernel.org/doc/html/v4.10/process/coding-style.html). I haven't looked too deeply at the code, so I'm not sure if it all conforms, but I saw no obvious issues.

berarma, 2021-03-29:
Thanks for sharing!

I was just starting to take it more seriously. I wanted to simplify the changes before starting commiting patches but I'm hesitating about talking to them first.

When you say touching base do you mean explaining them the changes without sending any code?

berarma, 2021-08-19:
I think #49 should be fixed before trying to upstream the driver.

motolav, 2023-01-31:
#49 is possibly fixed but needs someone with a G29 and Euro Truck Sim 1.41 to verify, You have to use the master tree as it was after version [0.4.0](https://github.com/berarma/new-lg4ff/releases/tag/0.4.0) where the issue might have been fixed as it fixed my issue with auto centering on the DFGT.

berarma, 2023-01-31:
> #49 is possibly fixed but needs someone with a G29 and Euro Truck Sim 1.41 to verify, You have to use the master tree as it was after version [0.4.0](https://github.com/berarma/new-lg4ff/releases/tag/0.4.0) where the issue might have been fixed as it fixed my issue with auto centering on the DFGT.

Nothing has been fixed in the driver, it's that ETS2 has changed the way it uses FFB.

I haven't tried to upstream the driver because, due to my limited time, it'll most probably be a lot of work for me to create the patches they'll want. And if they're picky, I have already received hints that they will, they could block the important parts because they're not a perfect implementation. The truth is that the driver implements the features games use but in corner cases it wouldn't do what it's supposed to do.

Given the high probability that I could do the work just to be rejected, I'll wait until someone close to the kernel people gives me an indication that they're really interested and tell me which parts should be improved before approaching upstream.

motolav, 2023-01-31:
> Nothing has been fixed in the driver, it's that ETS2 has changed the way it uses FFB.

I was testing with the OLD FFB version of ETS2 that's why i specified 1.41 as its the old FFB model, versions 1.42 and after use the new FFB model. There absolutely was an unintentional fix in the driver with the changes after 0.4.0.

> Given the high probability that I could do the work just to be rejected, I'll wait until someone close to the kernel people gives me an indication that they're really interested and tell me which parts should be improved before approaching upstream.

Submitting your code or someone else on your behalf to the mailing list is the only real way to actually get help on what needs to be changed

isopix, 2023-01-31:
Thats's right, even if  you are uninterested in doing all this work,
it may attract actual kernel dev, having this hardware and taking over
this process in the end. Logitech wheels are most popular BTW,

berarma, 2023-01-31:
I'll try to get in touch with the kernel maintainers and ask them what would I need to do to upstream it. Maybe it's not that hard.

berarma, 2023-01-31:
> I was testing with the OLD FFB version of ETS2 that's why i specified 1.41 as its the old FFB model, versions 1.42 and after use the new FFB model. There absolutely was an unintentional fix in the [...]

That's very weird. I'll have to try it.

motolav, 2023-01-31:
> I'll try to get in touch with the kernel maintainers and ask them what would I need to do to upstream it. Maybe it's not that hard.

Setting up an email client is the hardest part if you want people to be able to compile and test your code from an email. You will likely have to make a new repo/branch to create either a single large commit of all your changes or a few smaller commits of all your major changes. Sending your code as an [RFC] (Request for Comments) making a new branch/repo will be easier to review

motolav, 2023-02-03:
I pasted your driver into the kernel code see the diff and it seems a bit too much to submit as one commit, 1544 insertions(+), 531 deletions(-). (I manually updated hid-ids.h and hid-lg.c as there was a few changes after your fork)

howlett, 2023-06-14:
Don't worry about your code being bug-free before submitting, as long as it isn't going to break anything that works today.  You are adding functionality that is lacking.  There are totally bugs, when they are known, they can be fixed with patches.
Break your patches up as you said above, into logical units of work.  Make sure each one compiles cleanly.
Run scripts/checkpatch.pl against each one to ensure you have correct formatting.
Set up git to send email and use "git send-email" to submit them.  Cc the necessary lists (always lkml), and send it to the maintainer of the subsystem and anyone else that has contributed a lot to the code you are changing (use ./scripts/get_maintainer.pl to find out who)
Base your code on either Linus' branch or the subsystems git reop and tell the maintainer in the cover letter (git format-patch -<number of commits> --cover-letter, edit 000-cover-letter.patch and add details on the changes)

hzulla, 2023-11-05:
> Maybe it's not that hard.

Indeet, it's not. As a total beginner in kernel development, I contributed some minor HID patches and an insignificant gamepad device driver. The HID maintainers were friendly folks and very patient, contributing was a pleasant experience. Your driver is very helpful and would be a very good addition to the mainline kernel. Please be encouraged to go on upstreaming it. Thanks!

nyabinary, 2025-01-12:
Maybe @Lawstorant can help after upstreaming the universal-pidff stuff?

Lawstorant, 2025-02-24:
@berarma I must say, they might be more lenient than you would think, especially as force feedback is pretty niche in Linux. If you don't have some obvious bad code in the driver and you're keeping in the linux coding style, it should be fine

berarma, 2025-02-24:
> @berarma I must say, they might be more lenient than you would think, especially as force feedback is pretty niche in Linux. If you don't have some obvious bad code in the driver and you're keeping in the linux coding style, it should be fine.
> 
> Moreover, this is such a massive upgrade, that retracing your steps to split it into multiple commits could be not possible anymore? 

I read the rules to send a patch and was immediately taken aback.

Splitting would also mean throw away all the testing done for these years.

There are currently three points that I think should be worked before sending it:

- 32 bits compatibility.
- dynamic allocation of conditional effects.
- consider removing some experimental features?

I'm on mobile now, I'll put links to the related issues/PRs when I'm at the computer.

I will make an effort to work through this and try sending a patch.

Lawstorant, 2025-02-24:
Hmm, what were you taken aback with?

berarma, 2025-02-24:
> Hmm, what were you taken aback with?

Splitting the changes comes to mind.

hzulla, 2025-02-24:
Again, speaking from my own experience of contributing [a minor driver](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-bigbenff.c), the experience was a nice one. The HID maintainer was a friendly fellow and the review process helpful to me. I'd still want to encourage you to upstream the driver to the mainline kernel, since it would be an improvement for many gamers on Linux.

berarma, 2025-02-24:
> Again, speaking from my own experience of contributing [a minor driver](https://github.com/torvalds/linux/blob/master/drivers/hid/hid-bigbenff.c), the experience was a nice one. The HID maintainer [...]

I'm afraid they will see too much lines of code for something that they thought was already working. But as long as I don't have to split the changes into smaller commits I'm willing to try it.

hzulla, 2025-02-24:
The kernel maintainers usually are opinionated about what they accept and they may have different opinions than you do about how to implement things. It's your call if you want to engage in such a discussion and want to adapt your own code to their expectations. I found that discussion fruitful for my small contribution.

And I also like that my little driver now comes with any Linux device, making the force feedback gamepad work out of the box with all these retrogaming devices based on Linux. Also, other developers later improved my code, which is also nice.

hzulla, 2025-02-24:
> as long as I don't have to split the changes into smaller commits

The maintainers usually want to be able to understand what's going on. They ask to make commits small and each commit about a specific sub-feature or topic so that they can easily follow what you're doing in there.

berarma, 2025-02-24:
> The kernel maintainers usually are opinionated about what they accept and they may have different opinions than you do about how to implement things. It's your call if you want to engage in such a [...]

Maybe you haven't seen this code, but most of it is implementing the FFB effects by software. So basically I'm afraid that the kernel devs think it's not OK doing this in the kernel, or that the code is too complex for a driver. If I had to remove all that code there would be little left to upstream, except the LGPRO support.

hzulla, 2025-02-24:
Again, it's your call if you're really wanting to undergo the process.

I'm just encouraging you to do so since I see an advantage for users if your code ends up in mainline *and* and advantage for you as future maintenance will be on more shoulders than just yours, keeping the driver longer alive than you'll want to spend time taking care of it.

Maybe you can just send an email to the kernel driver mailing list, describe your driver, point to your github code and ask for opinions on if and how it's sensible to add this to mainline. Possibly there may be veteran HID developers willing to help you with the task of mainlining?

(Maybe, just maybe you can also ask the SteamOS folks if they are willing to sponsor the effort? This driver should be in their interest.)

berarma, 2025-02-24:
> Maybe you can just send an introductory email to the kernel driver mailing list, describe your driver, explain your design choices and point to your github code and then ask for opinions on if and how it's sensible to add this to mainline. Possibly there may be veteran HID developers willing to help you with the task of mainlining?

I've already sent that email and got no response.

I'm not saying I won't do it. I'll send everything in one patch when the issues I mentioned above are addressed and see what happens.