berarma/new-lg4ff #38: Support for shifting LEDs

miq, 2021-01-08:
It would be great, if the shifting LEDs of my G29 wheel would be supported.

What needs to be done here? Would gladly try to help out but have only very limited experience with driver development....

berarma, 2021-01-08:
What do you mean? They are currently supported.

Are you thinking about any game in particular?

miq, 2021-01-09:
Well, I play mostly ACC at the moment and the shifting/rev indicators on the wheel are off. I have no idea how to set them up. On windows it seems that fanaleds is the way to go...

berarma, 2021-01-09:
> Well, I play mostly ACC at the moment and the shifting/rev indicators on the wheel are off. I have no idea how to set them up. On windows it seems that fanaleds is the way to go...

That's a Windows game that you may be running on Wine/Proton. You can try the same solutions that would work on Windows and LEDSs need to be supported by Wine/Proton too.

In any case this is not a driver issue, it's a game issue, and we don't know if it's a Wine/Proton issue until tested with a game that supports the LEDs.

isopix, 2021-01-09:
That means that games doesn't support LEDs natively
Fanaleds is application that game device backends(that connect to them using different drivers) and game backends (that connects to game [usually by sockets] to get telemetry data from games and extract needed information. Sometimes they are some hacks, and sometimes it needs modding game to gain that]) You should search trough google. Usually it's easy to find info about game + g27/g29 leds + wine/proton/linux

miq, 2021-01-09:
Ok, thanks, will do some research and report the results here. I think we can close the issue then for documentation.

closed 2021-01-09 by berarma

d4rk, 2021-01-14:
I've written a tool to get shifting LEDs to work in Linux:

https://github.com/d4rk/ac_shifting_leds

Just run it before launching AC.

Cheers.