berarma/new-lg4ff #42: Steam doesn't see my steering wheel

markonius, 2021-02-13:
This likely isn't an issue with lg4ff, but I don't know where else to ask.  
I've installed lg4ff with `dkms` as instructed and the wheel is detecte in Oversteer, but it doesn't show up in Steam. It says "no controllers detected" and it doesn't show up in BeamNG.

What would be the next thing to check or try?

berarma, 2021-02-13:
How did you install Steam? Is it running in some form of container?

isopix, 2021-02-13:
Isn't that related to proton/wine build against SDL or evdev?

berarma, 2021-02-14:
> Isn't that related to proton/wine build against SDL or evdev?

It's about the detection of the device. you must be thinking about the FFB.

isopix, 2021-02-14:
Could you explain it?

I guess that both SDL and evdev can be used for detection, getting
joystick/controller data and setting FFB. And SDL uses evdev or old
joydev interface
So why wine needs to be compiled with SDL2 backend then. And if is it
superior backend, why it's not enabled by default (some wacom tablet
bugs or something?)

markonius, 2021-02-14:
Hi! Thanks for taking time to look into this.  
Steam is from rpmfusion-nonfree-updates, it's not a flatpak or anything. I have no trouble playing with a Steam Controller nor with a PS2 controller.  
I don't know what process to use to test what's wrong, to narrow down the issue. 

Just a random guess, could this be an issue with udev rules? I've never had the will to try and learn how they work.

berarma, 2021-02-14:
Assuming you're using the latest Proton versions. Make sure the wheel is plugged to the wall or it might fail.

There might be some setting in Steam that grabs the wheel as a controller, I'm not sure.

Udev permissions shouldn't be a problem. If the wheel works with Oversteer running as non-rooy user it should work everywhere else.

markonius, 2021-02-14:
Yeah, I don't know, it even works in Wine, force feedback and all. I'm gonna close the issue, lg4ff obviusly isn't the problem. Thanks anyway.

closed 2021-02-14 by markonius

markonius, 2021-02-14:
I figured it out, in case someone else stumbles upon this thread, I just had to disable steam input for the game.