berarma/new-lg4ff #43: Option to generate keyboard events for extra buttons

tnijkes, 2021-02-20:
Background info: F1 2020 on Google Stadia, or perhaps any other game on Google Stadia, does not support more than the standard 17 buttons on a gamepad-like device.
The extra buttons *do* work in the Chromium browser (checked with https://gamepad-tester.com), just not in Stadia.
I patched the Chromium browser to recognise the DFGT as a Standard gamepad device, but you could also use a browser extension like ControlStadia.

I have created a new module option in new-lg4ff to have the extra buttons generate regular keyboard events on my Driving Force GT.
That way, I can use *all* buttons on Google Stadia.

Would you consider adopting this option in the mainline?

berarma, 2021-02-20:
It's a hardcoded workaround for a Stadia issue aimed at a particular game. I don't think this fits the purpose of a driver module.

For the generic goal of mapping controllers to mouse/keys, I think sc-controller does a good work. I just wish it had better support for wheels but it can be used.

leillo1975, 2021-02-20:
Antimicro make this work perfectly:
https://github.com/AntiMicro/antimicro

berarma, 2021-02-20:
> Antimicro make this work perfectly:
> https://github.com/AntiMicro/antimicro

Great suggestion but you should use [AntiMicroX](https://github.com/AntiMicroX/antimicrox) instead by the Readme information.

closed 2021-02-20 by berarma

tnijkes, 2021-02-21:
> It's a hardcoded workaround for a Stadia issue aimed at a particular game. I don't think this fits the purpose of a driver module.
> 
> For the generic goal of mapping controllers to mouse/keys, I think sc-controller does a good work. I just wish it had better support for wheels but it can be used.

Yes, true, But isn't the kernel full of hardcoded workarounds? ;-)
I suggested to use a driver option to actually activate it, to ease the pain.

I have tried and used all other suggested workarounds, but such middleware isn't really an elegant option either.
Most importantly, they all introduce lag.

Well, I will keep on using my own ugly fork then :-)
Thanks.

jotomo, 2021-02-24:
I gave AntiMicroX a try to map some of the buttons from my G27 to control system volume, but Dirt Rally 2.0 seems to have exclusive access to the G27. 
My use-case would be controlling intercom volume by mapping G27 buttons to control system volume since the game doesn't directly support that adjusting volume.
@berarma Would a solution be technically feasible and appropriate for a driver that uses a module parameter with a list of _buttonId -> keySymbol / keyId_ mappings that can be specified to cover use-cases for all wheel and games?

berarma, 2021-02-24:
> I gave AntiMicroX a try to map some of the buttons from my G27 to control system volume, but Dirt Rally 2.0 seems to have exclusive access to the G27.

It would be a very unusual case. Have you tried running Oversteer and using it while the game is running? I guess it will work.

> My use-case would be controlling intercom volume by mapping G27 buttons to control system volume since the game doesn't directly support that adjusting volume.
> @berarma Would a solution be [...]

I don't think it's a good idea at all. It would make a lot more sense, and I guess it would be more beneficial for all, to figure out why AntiMicroX isn't working and try to fix it. Let's fix the hammer instead of trying to make the screwdriver behave as a hammer.

jotomo, 2021-02-24:
> I don't think it's a good idea at all

I had a hunch you'd say that ;-) I'll give AntiMicroX another try, I might be thinking of Win7 behavior w.r.t. to exclusive access. If AntiMicroX can handle the use-case I certainly agree it should go there, rather than a driver.