berarma/new-lg4ff #44: G923 support

Elizafox, 2021-03-29:
Hello,

I noticed this driver doesn't work with the G923, which also doesn't work with stock linux. 😞

What does work with stock linux:
* The buttons and paddle shifters (most of them at least)
* The directional buttons, detected as a joystick with four axes (but only on or off)

What doesn't work:
* The wheel
* The pedals
* +/-, enter, and the click wheel

Any luck that this could be supported? I have this wheel and a little kernel programming experience, although not in HID devices.

There is [this project](https://github.com/ZRtmWrJqXcjbqBLIMBYMCeUw/Logitech-G923-Linux-Kernel-Driver/), but they don't appear to have gotten very far.

berarma, 2021-03-29:
Hi.

Sorry, this isn't supported and I can't do anything about it. I'd like to help but I don't have the means.

I don't know very much about HID devices neither, I just modified the module so that it could process more effect types and other changes that had little to do with the HID interface.

Our best bet would be that the wheel could be switched to a mode compatible with the G29 or G923. But we'd need to know how to do it.

Logitech had released information about all their previous models. Maybe if we contacted the right people at Logitech they would release some programming docs.

isopix, 2021-03-29:
It doesn't sound so bad, at least you can use it as big and pricey
(and uncomfortable) joypad.

But seriously you can sell it and buy two used G29 or G27 for that
price. Or much more better(but less reliable) Fanatec/Thrustmaster
gear for similar price.

If you look over ticket's on this github(or forks) you will find that
some devs tried, but was unfortunate to do it

G29 got all the knobs you got on your G923, but I still recommend G27
with buttons in much more stable position(on shifter, that doesn't
rotate like wheel).
You shouldn't have problem to sell G923 for 80%-90% of new unit on
sale(they are usually on sell, so I hope you haven't paid full price
;-)
G29 is PC/PS3/PS4/PS5 compatible, so it's still good buy if you need
console compatiblity(and resellability, because of such wide
compatiblity).

But because you wrote you've got some kernel programming experience,
and if you are not in hurry to use your gear(play games), then you can
have more fun hacking protocol and finding magic sequence to
reintialize it(you must prepare windows machine with WireShark or some
VM wih usb dump ability. Like I wrote, there already were unsuccesfull
attempts to do it, so let's start with finding them.

anyway good luck with reselling it or tinkering for weeks(or months or
even years ;-)

Elizafox, 2021-03-30:
> It doesn't sound so bad, at least you can use it as big and pricey (and uncomfortable) joypad. But seriously you can sell it and buy two used G29 or G27 for that price. Or much more better(but less [...]

It presents as 3 hidraw devices (which doesn't seem right to me). The first one seems to be the one that works. I can get data when I rotate the wheel when I run hexdump on it, and although I see some patterns I've not yet decoded what it all means yet.

P.S. there really is no need to be insulting (really — I don't appreciate it, and if that's how you treat your contributors, I'll happily go do this on my own and stop wasting your time), and no, I have no intention of selling my wheel.

isopix, 2021-03-31:
Sorry if you feel insulted, but I don't get why (and it wasn't my intention).
I simply wrote that development start for G923 and were nowhere and
that if you want usable device now, then it's better to sell and buy
much cheaper G29 or G27(you can buy two units for G923 price), or feel
free to hack it for weeks, months or longer(seeing how current
development efforts stalled).
On the bright side seeing also as development went(few months for
working drivers) for totally different Fanatec and Thrustmaster gear
went, it should be theoretically easier for G923, which at least
mechanically is identical to G29

Feel free to point me at part of message where you felt insulted,
because I cannot read my previous that way.
And I'm not related to Berarma(dev of this project) at all. I'm simply
G29 owner that follows wheel drivers development(for different
wheels), so my message is definitely "not how new-lg4ff developers
threat controibutors"

berarma, 2021-08-19:
There's already a PR to add support for this wheel.

closed 2021-08-19 by berarma