berarma/new-lg4ff #47: G25 SDL_HAPTIC_CONSTANT effect, direction not correct on Raspberry PI 4

Fredobedo, 2021-05-17:
Hi Dear,
I request you assistance as I think there may be a bug in new-lg4ff.
When I install new-lg4ff module driver on my PI4, I remark that it is not possible anymore to execute a constant effect on my Logitech G25, with direction to right (-1); it always react as left.
The same code on the same machine with default drivers is working well.

SDL version is 2.0.14 (but I also tested with previous version and had the same problem).

Here is an extract of my test code:
<<
...
void TriggerConstantEffect(int direction, double strength)
{
	printf("TriggerConstantEffect-direction=%d\n", direction);

	debug(1, "TriggerConstantEffect\n");
	if (supported & SDL_HAPTIC_CONSTANT) 
	{
		SDL_HapticEffect tempEffect;
		SDL_memset(&tempEffect, 0, sizeof(SDL_HapticEffect));
		tempEffect.type = SDL_HAPTIC_CONSTANT;
		tempEffect.constant.direction.type = SDL_HAPTIC_CARTESIAN;
		tempEffect.constant.direction.dir[0] = direction;
		tempEffect.constant.direction.dir[1] = 0;
		tempEffect.constant.direction.dir[2] = 0;
		tempEffect.constant.length = SDL_HAPTIC_INFINITY;
		tempEffect.constant.delay = 0;
		tempEffect.constant.level = level;

		if(SDL_HapticUpdateEffect(haptic, effects.effect_constant_id, &tempEffect)!=0)
			debug(1, "Error updating effect: %s\n", SDL_GetError());
		else if(SDL_HapticRunEffect(haptic, effects.effect_constant_id, SDL_HAPTIC_INFINITY)!=0)
			debug(1, "Error executing effect: %s\n", SDL_GetError());
		else
			debug(1,"->success\n");	
		sleep(5);
	}
}
...
>>
 
Thank you in advance for your support
Fred

berarma, 2021-05-17:
Please, can you provide a simple compilable test that shows the issue?

I've used your test code by filling the gaps and it works correctly?

Fredobedo, 2021-05-17:
Thank you to reply so fast!
Did you try on a Raspberry, please?
Because my code is working correctly on my dev laptop with Ubuntu, the problem occurs only on my PI4.

I will prepare you the code this afternoon
Fred

Fredobedo, 2021-05-17:
Here is a sample code, I just tested it on my PI and I confirm it's turning to the left.
if I set direction to 0, it also turns to left.

I compiled it like this:
"gcc testHaptic.c -o testHaptic -lSDL2"
[testHaptic.c.txt](https://github.com/berarma/new-lg4ff/files/6493429/testHaptic.c.txt)

Thank you
Best regards

berarma, 2021-05-17:
> Did you try on a Raspberry, please?

No. I can't try on RPi.

I can only think of some compilation issue. Is it a 32 or 64 bits system?

Fredobedo, 2021-05-17:
Dear,
I did the test again and confirm that setting direction to 0 and 1, both move the wheel to the left.

It's a 32bit distribution:
--------------------------
Distributor ID : Raspbian
Description    : Raspbian GNU/Linux 10 (buster)
Release        : 10
Codename       : buster
kernel release : 5.10.17-v7l+
Processor      : ARMv7 Processor rev 10 (v7l)

berarma, 2021-05-18:
I've created a PR that might fix your problem. Can you try it please?

Let's move the conversation there.

PR #48 (2021-06-07): Fix types for 32bits systems
Ensure that 64bits variables are used where is needed even on 32bits systems. This should fix #47.
closed 2021-05-18 by berarma