berarma/new-lg4ff #55: FFB sometimes doesn't work on subsequent game starts until wheel is re-connected

Espionage724, 2021-11-08:
If I start up City Car Driving, FFB is fine. But if I close it and go to Project CARS 3, there is no FFB/resistance (was the reason for https://github.com/berarma/new-lg4ff/issues/54). If I unplug the wheel and re-connect it, FFB/resistance works fine in PCARS 3.

berarma, 2021-11-08:
It seems this is some issue specific to that game, I don't think it's related to this driver. Can you confirm?

Espionage724, 2021-11-13:
> It seems this is some issue specific to that game, I don't think it's related to this driver. Can you confirm?

It does seem specific to PCARS 3. If I start City Car Driving (CCD), close it, and re-open it, FFB still works fine on it.

Any suggestions to figure out why PCARS 3 would require a wheel hotplug at times?

berarma, 2021-11-13:
You could try Oversteer and see if the game is messing with the FFB global gain.

If that doesn't work, you can use ffbtools to output a log where we could try to guess what's happening.

It should be reported to Proton since this will probably be a Proton issue and will have to be fixed there.

berarma, 2021-11-13:
Are you using ProtonGE or vanilla Proton?

Espionage724, 2021-11-13:
Thanks, I'll give Oversteer and ffbtools a try.

> Are you using ProtonGE or vanilla Proton?

Vanilla Proton (Experimental)

closed 2022-09-01 by berarma