berarma/new-lg4ff #57: module won't compile w/ kernel 4.18.0-348.7.1.el8_5.x86_64

6161726f6e, 2022-01-22:
Hi, getting this error with kernel 4.18.0-348.7.1.el8_5.x86_64:
`# dkms install /usr/src/new-lg4ff`
Deprecated feature: REMAKE_INITRD
Deprecated feature: REMAKE_INITRD
Creating symlink /var/lib/dkms/new-lg4ff/0.3.3/source -> /usr/src/new-lg4ff-0.3.3

Building module:
cleaning build area...
make -j16 KERNELRELEASE=4.18.0-348.7.1.el8_5.x86_64 KVERSION=4.18.0-348.7.1.el8_5.x86_64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.18.0-348.7.1.el8_5.x86_64 (x86_64)
Consult /var/lib/dkms/new-lg4ff/0.3.3/build/make.log for more information.

-------------------------------------------

`# cat /var/lib/dkms/new-lg4ff/0.3.3/build/make.log`
DKMS make.log for new-lg4ff-0.3.3 for kernel 4.18.0-348.7.1.el8_5.x86_64 (x86_64)
Sat Jan 22 12:45:17 EST 2022
make -C /lib/modules/4.18.0-348.7.1.el8_5.x86_64/build M=$PWD
make[1]: Entering directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
  CC [M]  /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg.o
  CC [M]  /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.o
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:1043:5: error: redefinition of ‘lg4ff_adjust_input_event’
 int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:22:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.h:15:19: note: previous definition of ‘lg4ff_adjust_input_event’ was here
 static inline int lg4ff_adjust_input_event(struct hid_device *hid, struct hid_field *field,
                   ^~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:1069:5: error: redefinition of ‘lg4ff_raw_event’
 int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
     ^~~~~~~~~~~~~~~
In file included from /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:22:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.h:17:19: note: previous definition of ‘lg4ff_raw_event’ was here
 static inline int lg4ff_raw_event(struct hid_device *hdev, struct hid_report *report,
                   ^~~~~~~~~~~~~~~
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c: In function ‘lg4ff_alternate_modes_store’:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:1545:60: error: ‘lg4ff_no_autoswitch’ undeclared (first use in this function); did you mean ‘lg4ff_compat_mode_switch’?
  if (target_product_id == USB_DEVICE_ID_LOGITECH_WHEEL && !lg4ff_no_autoswitch) {
                                                            ^~~~~~~~~~~~~~~~~~~
                                                            lg4ff_compat_mode_switch
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:1545:60: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c: In function ‘lg4ff_handle_multimode_wheel’:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:2070:7: error: ‘lg4ff_no_autoswitch’ undeclared (first use in this function); did you mean ‘lg4ff_compat_mode_switch’?
      !lg4ff_no_autoswitch) {
       ^~~~~~~~~~~~~~~~~~~
       lg4ff_compat_mode_switch
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c: At top level:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:2095:5: error: redefinition of ‘lg4ff_init’
 int lg4ff_init(struct hid_device *hid)
     ^~~~~~~~~~
In file included from /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:22:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.h:19:19: note: previous definition of ‘lg4ff_init’ was here
 static inline int lg4ff_init(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:2306:5: error: redefinition of ‘lg4ff_deinit’
 int lg4ff_deinit(struct hid_device *hid)
     ^~~~~~~~~~~~
In file included from /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.c:22:
/var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.h:20:19: note: previous definition of ‘lg4ff_deinit’ was here
 static inline int lg4ff_deinit(struct hid_device *hdev) { return -1; }
                   ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:316: /var/lib/dkms/new-lg4ff/0.3.3/build/hid-lg4ff.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1571: _module_/var/lib/dkms/new-lg4ff/0.3.3/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-348.7.1.el8_5.x86_64'
make: *** [Makefile:5: default] Error 2

--------------------------------------------------------------

Any guidance would be appreciated.

Thanks.

berarma, 2022-01-22:
Your kernel has to be compiled with `CONFIG_LOGIWHEELS_FF`.

6161726f6e, 2022-01-22:
Ahhhh.  OK.  Thanks!

6161726f6e, 2022-01-22:
Do you know if I should also include `LOGITECH_FF`?

6161726f6e, 2022-01-22:
Do the following config options look good?

CONFIG_HID_UCLOGIC=m
CONFIG_HID_LOGITECH=m
CONFIG_HID_LOGITECH_DJ=m
CONFIG_HID_LOGITECH_HIDPP=m
# CONFIG_LOGITECH_FF is not set
# CONFIG_LOGIRUMBLEPAD2_FF is not set
# CONFIG_LOGIG940_FF is not set
CONFIG_LOGIWHEELS_FF=y

berarma, 2022-01-23:
> Do you know if I should also include `LOGITECH_FF`?

I guess it's not needed.

berarma, 2022-01-23:
> Do the following config options look good?
> 
> CONFIG_HID_UCLOGIC=m
> CONFIG_HID_LOGITECH=m
> CONFIG_HID_LOGITECH_DJ=m
> CONFIG_HID_LOGITECH_HIDPP=m
> \# CONFIG_LOGITECH_FF is not set
> \# CONFIG_LOGIRUMBLEPAD2_FF is not set
> \# CONFIG_LOGIG940_FF is not set
> CONFIG_LOGIWHEELS_FF=y
> 

Yes.