berarma/new-lg4ff #59: SnowRunner FFB problems

tepsaj, 2022-03-20:
I was trying to get my Logitech G29 wheel to work with [SnowRunner](https://store.steampowered.com/app/1465360/SnowRunner/), but I'm having problems with it. 

First of all I tested that my G29 works with ETS2, so the new-lg4ff installation should be correct.

Using Proton 7.0-1 and 6.3-8 I was able to get the SnowRunner game recognize the wheel using `-e` flag in steam launch arguments. In the game settings the wheel is recognized when the steering wheel API is set to either DirectInput or Thrustmaster. Setting it to Logitech Legacy or Logitech Trueforce doesn't detect the wheel. When I have been playing the game with Windows 10, the Logitech Legacy has been the correct option.

I can map the buttons and use the wheel to steer but the FFB isn't working properly. I don't have any FFB when turning the wheel but sometimes it applies a huge spike of FFB, I think. This lasts maybe a half second and makes a pretty bad noise from the wheel base. 

I'm ready to test and debug the problem further, if you can give me hints what to look for and what tools to use. I'm new to the whole world of Linux gaming and getting this game to work with the wheel would be fantastic!

berarma, 2022-03-20:
This isn't related to the driver, you should report it to Proton developers. There should be an issue dedicated to that game.

Proton 7.0 doesn't work well with wheels, better stay with 6.3-8.

I guess the API should be DInput since there's no propietary APIs implemented in Proton as far as I know.

tepsaj, 2022-03-20:
Okay, Thanks! I will continue with the Proton developers.

closed 2022-03-20 by tepsaj