berarma/new-lg4ff #64: Rumble effect keeps adding up

Norodix, 2022-09-17:
Hey,
Ive been using this driver for some time with my Logitech G25. Now I'm playing Assetto Corsa and it seems to me there is a strange issue. As the laps go on it feels like there is a constantly increasing rumble effect. It happens when going in a straight line. It reacts to my speed, so the effect slows down when I'm going slowly and increases in frequency when going fast. But it feels like its stronger from lap to lap.

Can you help me how I can debug what is happening? Is there a way to read out the currently applied effects and its parameters from the driver, or the calls to the driver?

System:
  Kernel: 5.19.0-76051900-generic x86_64 bits: 64 compiler: N/A
    Desktop: GNOME 42.3.1 Distro: Pop!_OS 22.04 LTS
    base: Ubuntu 22.04 LTS Jammy

berarma, 2022-09-17:
Interesting. You can use ffbtools as a wrapper to log ffb commands. You can share the log here.

Norodix, 2022-09-17:
It doesnt seem to log calls to the driver.
This is all I get in the logfile, while the ffb is wokring.
`000000000000 # DEVICE_NAME=046d G25 Racing Wheel, UPDATE_FIX=1, DIRECTION_FIX=0, FEATURES_HACK=1, FORCE_INVERSION=0, IGNORE_SET_GAIN=0, OFFSET_FIX=0, THROTTLING=0`

This is my command:
`/home/bence/Downloads/ffbtools/bin/ffbwrap --update-fix --logger=/home/bence/Downloads/ffbtools/logs.txt --features-hack /dev/input/js0 -- %command%`

berarma, 2022-09-17:
The device isn't /dev/input/js0. You have to look for it into /dev/input/by-id and it's the evdev entry.

Norodix, 2022-09-18:
Okay, thanks for the help, I managed to make the logs. 

Reading into it I didnt see anything so I tried to experiment a bit more and it turns out it was a new-to-me effect in assetto corsa that was caused by flattening my tires. I used to have abs turned on but this time when I picked up I decided to keep it off. I changed hardware and sw in the meantime that is why I suspected a sw issue, but it was PEBKAC. Thanks for the help figuring it out!

closed 2022-09-18 by Norodix