berarma/new-lg4ff #65: No effects except autocenter.

frank038, 2022-10-21:
I've installed this driver to be used with wine and rFactor1.
Except for the autocenter effect, all the others are missed. For example, no effects while driving over curbs.
OS: Debian testing (next stable) - kernel 5.19
Wine version: 7.19 (I don't think it is related to a specific wine version because the same issues happen with Proton 7.0-15).

berarma, 2022-10-21:
Can you try with Proton 6.3-8? There were breaking changes in Wine/Proton 7.

frank038, 2022-10-21:
> Can you try with Proton 6.3-8? There were breaking changes in Wine/Proton 7.

Please, I can only use Wine/Wine Staging (from version 6.16 to 6.23) or GE Proton.
At the moment, I'm going to try GE Proton 6.4-GE-1.

berarma, 2022-10-21:
Any Wine 6.x should work. Proton GE is pretty unstable when it comes to wheel devices.

Can you test FFB with any other application not using Wine/Proton? Like Oversteer or some native game?

frank038, 2022-10-22:
Tested:
Wine staging 6.23: failed.
Proton-6.4-GE-1: success.
Thank you.

berarma, 2022-10-22:
Ok. This is definitely a Wine issue.

closed 2022-10-28 by berarma