berarma/new-lg4ff #68: new G29 usb id

ZyanKLee, 2022-12-01:
Hello and thanks a lot for creating and maintaining this great project.

Today I received my new G29 which did not work from the start. After some debugging I found it to have a new USB ID:

```
Bus 003 Device 012: ID 046d:c260 Logitech, Inc. G29 Driving Force Racing Wheel [PS4]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x046d Logitech, Inc.
  idProduct          0xc260 G29 Driving Force Racing Wheel [PS4]
  bcdDevice           89.00
  iManufacturer           1 Logitech
  iProduct                2 G29 Driving Force Racing Wheel
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     160
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
Device Status:     0x0000
  (Bus Powered)
```

After simply replacing the old G29 ID with my new one in my local clone of `new-lg4ff/hid-ids.h`, the module compiles fine and when loading it, dmesg finally prints the long awaited messages about ForceFeedback and version 0.4.0:

```
[Do Dez  1 15:31:34 2022] usb 3-6.1: new full-speed USB device number 12 using xhci_hcd
[Do Dez  1 15:31:34 2022] usb 3-6.1: New USB device found, idVendor=046d, idProduct=c260, bcdDevice=89.00
[Do Dez  1 15:31:34 2022] usb 3-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Do Dez  1 15:31:34 2022] usb 3-6.1: Product: G29 Driving Force Racing Wheel
[Do Dez  1 15:31:34 2022] usb 3-6.1: Manufacturer: Logitech
[Do Dez  1 15:31:34 2022] input: Logitech G29 Driving Force Racing Wheel as /devices/pci0000:00/0000:00:01.2/0000:20:00.0/0000:21:08.0/0000:2a:00.3/usb3/3-6/3-6.1/3-6.1:1.0/0003:046D:C260.0013/input/input51
[Do Dez  1 15:31:34 2022] logitech 0003:046D:C260.0013: input,hiddev2,hidraw3: USB HID v1.10 Joystick [Logitech G29 Driving Force Racing Wheel] on usb-0000:2a:00.3-6.1/input0
[Do Dez  1 15:31:34 2022] logitech 0003:046D:C260.0013: Force feedback support for Logitech Gaming Wheels (0.4.0)
[Do Dez  1 15:31:34 2022] logitech 0003:046D:C260.0013: Hires timer: period = 2 ms
[Do Dez  1 15:31:35 2022] usb 1-2.3: reset high-speed USB device number 6 using xhci_hcd
```

ZyanKLee, 2022-12-01:
aww man, just now saw the hint in oversteer issues, sorry to bother you!

for everybody coming here with the same issue, read this: https://github.com/berarma/oversteer/issues/84#issuecomment-982888082

closed 2022-12-01 by ZyanKLee

berarma, 2022-12-01:
Glad you figured it out so fast.

I'll put a note in the readme. I thought everybody would already know because it's a requirement for the in-kernel module to work but it's not clearly advertised anywhere.