berarma/new-lg4ff #73: release a build version?

omnom143, 2023-02-10:
its really difficult and i really dont want to troubleshoot for 3 hours just to get a driver that likely doesn't even work ive ran into a ton of errors

leillo1975, 2023-02-10:
It's really not that difficult. If you follow the README in 5 minutes it is installed. By the way, the driver works very well, without errors and in ALL car games, so I don't understand why you say that.

berarma, 2023-02-10:
It would be really difficult because the module built for a kernel version wouldn't work on another version. I would have to make as many builds as kernel versions there are and it still wouldn't work for custom versions that might be shipped on some distros. It's impractical.

The procedure is almost as easy as it would be installing a built module. And if you find issues we can help.

omnom143, 2023-02-10:
> It's really not that difficult. If you follow the README in 5 minutes it is installed. By the way, the driver works very well, without errors and in ALL car games, so I don't understand why you say that.

it is difficult when you dont give any insigt on how to install the tools and use the tools, its a general explanation not a step by step

berarma, 2023-02-10:
The only steps that aren't detailed are:

- Installing DKMS because it's dependant on the distro used. You should know how to install a package from your distro's repositories.
- Downloading the project files because there are several methods, the easiest of them it's probably downloading a release file and uncompressing it.

For these steps you only need to know the basics to use a computer and your distro specifically.

Then it's just a matter of executing one command that is detailed in the README.

Where are you having issues?

omnom143, 2023-02-10:
> The only steps that aren't detailed are:
>
> * Installing DKMS because it's dependant on the distro used. You should know how to install a package from your distro's repositories.
> * Downloading the [...]

all the programs you had me get didnt exist in pacman

Norodix, 2023-02-10:
https://archlinux.org/packages/extra/any/dkms/
Is this not it?

berarma, 2023-02-10:
> all the programs you had me get didnt exist in pacman

Which distro are you using? It's hard to believe there's some distro that doesn't offer `dkms`.

omnom143, 2023-02-10:
> Which distro are you using? It's hard to believe there's some distro that doesn't offer `dkms`.

arch based steam os

isopix, 2023-02-10:
If it's Steam Deck, then it's get complicated, because SteamInput is
breaking many things for wheels(but Old Momo wheel worked in desktop
mode).

About ArchLinux, lg4ff is in AUR:
https://aur.archlinux.org/packages/new-lg4ff-dkms-git

The esiest way to install it, would be switcing rootfs in SteamOS to
RW, and then you can install it from AUR (you must build package):

https://help.steampowered.com/en/faqs/view/671A-4453-E8D2-323C
"sudo steamos-readonly disable"

https://steamdecktips.com/blog/install-archlinux-packages-on-the-steam-deck

There are best tips IMHO:
https://steamcommunity.com/app/1675200/discussions/0/3395176266124615154/

closed 2023-07-12 by berarma