berarma/new-lg4ff #76: Struggling with G923 PS4/PC

JustinTJones, 2023-05-25:
Running Debian (sid) with Linux kernel 6.1.0-9.

I attempted the dkms version of installing this driver and updated initramfs as described in the README. It didn't seem to change how the wheel was being interpreted.

Then I read on the Arch forums that Debian based systems worked better with the manual install method. So I followed the instructions provided in this repo for manually building and installing. dmesg output doesn't seem to be any different from using the default driver other than listing logitech as the driver instead of hid-generic. See screenshot below. I unplugged the wheel and plugged it back in to see if it helped and to give myself a reference point in dmesg. Screenshot from time it was plugged back in.

[image]

I have another screenshot of dmesg output during evtest testing after a manual install of this driver, followed by manually unloading the driver and reloading it, then followed by the unplug, re-plug, and what is included above. 

[image]

I greatly appreciate the work that has been and continues to be put into this driver. 

I feel like I'm missing a step somewhere in getting this installed correctly, though I followed the instructions as written.

What can I do to help? Because I'm about to the point of putting my steering wheel back into a box in the closet for another couple of years.

JustinTJones, 2023-05-25:
I tried connecting the wheel directly to my computer, instead of using a usb hub and it worked. Thanks!

Unfortunately after testing it with the usb hub and then plugging it back into the computer, the wheel fails to rotate completely during it's startup sequence and is not even seen by the computer at all. Have you ever experienced this issue?

berarma, 2023-05-25:
No. Try different USB ports. What's the output of `lsusb`?

JustinTJones, 2023-05-25:
I had taken a break from it earlier and had not run lsusb at that time. It happened again a few minutes ago and lsusb hung for a bit, then when it finally finished listing devices the wheel was included and then started to work. The wheel is now working as expected in evtest and in oversteer. I'm going to fire up some games and see how we go.

Thank you so much for taking the time to reply and for creating this driver. I'm looking forward to finding other ways to contribute to the project.

closed 2023-05-25 by JustinTJones