berarma/new-lg4ff #78: the module seems to break hid-steam

alxwk, 2023-06-16:
If I try to use Steam Controller after using a wheel (so when hid-logitech-new is loaded into memory) I after some time experience random disconnects and freezes of the controller. Unloading this module first seems to fix the issue. Maybe, some timing settings go off or something.

(BTW, this dual naming "hid-logitech/hid-logitech-new" is problematic in itself. I can't unload the module with `modprobe -r`, as it can't find "hid-logitech-new", and when using `modprobe -r hid-logitech` it silently ends, but the module still stays in `lsmod` output. Only `rmmod hid-logitech-new` does the trick.)

berarma, 2023-07-12:
I don't think this module can affect the Steam Controller. It's the only report and it will probaby be hard to reproduce. I'll need some help.

The reason for the different name I think was to tell them apart when loaded. I use `rmmod hid-logitech-new`, I don't know why it doesn't work with `modprobe -r`. I'll do some tests.

alxwk, 2023-07-16:
Hmmm... Actually I just noticed weird Steam Controller behaviour even without `hid-logitech-new` loading since reboot. It might happen to be some unrelated issue, maybe even hardware one, and its happening after a wheel being connected is just a coincidence...

berarma, 2025-02-05:
Please, reopen if needed.

closed 2025-02-05 by berarma