berarma/new-lg4ff #80: Mouse clicking when FF events occur

nww02, 2023-07-04:
I have a Logitech Driving Force GT, playing Snowrunner (DirectInput) on Steam (proton 8.0.2) on Linux Mint 21.1

The wheel worked fine on the stock logitech-hid, but without Force Feedback.
The wheel works fine on the new lg4ff, and now has Force Feedback! yay!

HOWEVER  for some reason, when the game is issuing FF commands to the wheel (judder or just the engine rumble of idle), it causes the mouse cursor to issue 'left-clicks',  so if I'm stopped and look at the map, say, the map keeps being clicked.

Any idea why this might be happening????

berarma, 2023-07-04:
I don't think it's related to the Logitech driver.  Maybe some hardware interference or some other hardware issue. It could be something more exotic like the wheel making the table vibrate and affecting the mouse.

Can you try different USB ports and/or different mouse?

nww02, 2023-07-04:
I am wondering if it might be a bad interaction between Snowrunner and the driver.   When the game issues an impulse to the Force Feedback, it causes the cursor to issue a 'click', and on the map it occurs in the centre of the screen.  

I've just tried to reinstall lg4ff and rebooted, and now the wheel is just constantly moving left and right at maximum force.  

I'm going to try ETS2 and see if it's something with my system or SnowRunner.

berarma, 2023-07-04:
> It's definitely a problem between SnowRunner and the driver.  I just tried Euro Truck Sim 2 and it worked exactly as you'd expect.   
> 
> I did mod unload and modprobe showed hid-logitech back in the action, and SnowRunner just went berzerk, throwing the wheel left and right. Paradoxically, with the default driver loaded, ETS2 worked pretty well, FF included.  
> 
> So, the FF in the default driver really disagreed with SnowRunner.
> 
> So, I re-loaded lg4ff, and again, both SnowRunner and ETS2 work fine, but again Snowrunner is constantly doing a silent "mouse click" at the centre of the screen whenever the FF activates with lg4ff loaded.  Very strange... ETS2, though, works fine,.
> 
> 
> 

It's strange. If it works well with other software I guess the problem is in Proton.

nww02, 2023-07-04:
I wish I had the knowledge to track it down, but it's all a bit of a mystery.    Thank you for  lg4ff, it's a brilliant piece of software and helps me kick the Windows habit :)

nww02, 2023-07-04:
As it's an incompatibility with a particular game, it's likely the game itself is using the driver in an unexpected way, and/or doing something funny via proton.   Doesn't sound like lg4ff can help, so closing with thanks to the developer :)

closed 2023-07-04 by nww02