berarma/new-lg4ff #85: Can't make it load automatically

Chrom25, 2023-11-22:
Hello! 
so I have installed the driver using dkms without any issues and if I load it manually by using 
`
 - rmmod hid-logitech
 - modprobe hid-logitech
 `
it work as expected, but I can't make it load automatically after reboot on Ubuntu 23.10. 
I did 

`sudo update-initramfs -u (and even sudo update-initramfs -c -k $(uname -r)) `

but after reboot lsmod steel shows me 
`
ff_memless             24576   1 hid_logitech
hid_generic            12288    0
usbhid                     77824    1 hid_logitech
hid                           180224   3 hid_logitech,usbhid,hid_generic
`

so each time after reboot I need to do rmmod modprobe to make it work.
Any thoughts?

PS. on Fedora it works after dracut --regenerate-all without any issue.

berarma, 2023-11-22:
Can you paste the content of the following files?

/etc/initramfs-tools/initramfs.conf
/etc/initramfs-tools/update-initramfs.conf
/etc/initramfs-tools/modules

razcore-rad, 2025-10-06:
I installed via `dkms` a few days ago and I think I can confirm this. I found this in `sudo dmesg`:
```
[  434.226152] hid_logitech_new: module verification failed: signature and/or required key missing - tainting kernel
```
And couldn't found any reference to `logitech` with `lsmod`. But manually loading the module with `sudo modprobe hid-logitech` does indeed show it with `lsmod`:
```
         STDIN
     1   Module                  Size  Used by
*    2   hid_logitech_new       81920  0
*    3   ff_memless             24576  1 hid_logitech_new
[...]
*  156   usbhid                 77824  1 hid_logitech_new
*  157   hid                   266240  13 usbhid,hid_generic,hid_logitech_new,uhid
```
Here are my `initramfs-tools` files. I don't see any reference to `logitech` in them:
```
# /etc/initramfs-tools/initramfs.conf

#
# initramfs.conf
# Configuration file for mkinitramfs(8). See initramfs.conf(5).
#
# Note that configuration options from this file can be overridden
# by config files in the /etc/initramfs-tools/conf.d directory.

#
# MODULES: [ most | netboot | dep | list ]
#
# most - Add most filesystem and all harddrive drivers.
#
# dep - Try and guess which modules to load.
#
# netboot - Add the base modules, network modules, but skip block devices.
#
# list - Only include modules from the 'additional modules' list
#

MODULES=most

#
# BUSYBOX: [ y | n | auto ]
#
# Use busybox shell and utilities.  If set to n, klibc utilities will be used.
# If set to auto (or unset), busybox will be used if installed and klibc will
# be used otherwise.
#

BUSYBOX=auto

#
# COMPRESS: [ gzip | bzip2 | lz4 | lzma | lzop | xz | zstd ]
#

COMPRESS=zstd

#
# COMPRESSLEVEL: ...
#
# Set a compression level for the compressor.
# Defaults vary by compressor.
#
# Valid values are:
# 1 -  9 for gzip|bzip2|lzma|lzop
# 0 -  9 for  lz4|xz
# 0 - 19 for zstd
#
# COMPRESSLEVEL=1

#
# DEVICE: ...
#
# Specify a specific network interface, like eth0
# Overridden by optional ip= or BOOTIF= bootarg
#

DEVICE=

#
# NFSROOT: [ auto | HOST:MOUNT ]
#

NFSROOT=auto

#
# RUNSIZE: ...
#
# The size of the /run tmpfs mount point, like 256M or 10%
# Overridden by optional initramfs.runsize= bootarg
#

RUNSIZE=10%

#
# FSTYPE: ...
#
# The filesystem type(s) to support, or "auto" to use the current root
# filesystem type
#

FSTYPE=auto
```

```
# /etc/initramfs-tools/update-initramfs.conf

#
# Configuration file for update-initramfs(8)
#

#
# update_initramfs [ yes | all | no ]
#
# Default is yes
# If set to all update-initramfs will update all initramfs
# If set to no disables any update to initramfs beside kernel upgrade

update_initramfs=yes

#
# backup_initramfs [ yes | no ]
#
# Default is no
# If set to no leaves no .bak backup files.

backup_initramfs=no
```

```
# /etc/initramfs-tools/modules

# List of modules that you want to include in your initramfs.
# They will be loaded at boot time in the order below.
#
# Syntax:  module_name [args ...]
#
# You must run update-initramfs(8) to effect this change.
#
# Examples:
#
# raid1
# sd_mod
```