berarma/new-lg4ff #86: G923 Trueforce going crazy in EA WRC

eruizc-dev, 2023-11-24:
When playing EA WRC I noticed a weird behavior, after turning the wheel fast enough it will start going left and right until I stop it around the middle (see the video below). This does not happen on menus, only when sitting in a car.

https://github.com/berarma/new-lg4ff/assets/58179604/30a6962d-9cd9-411e-9f38-4dc052f7b788

This is very annoying as it can happen during a race, I've been using this exact setup for months in other games without a problem. I've played Dirt Rally, Dirt Rally 2.0, and Assetto Corsa Competizione all in which force feedback works flawlessly. In fact I played a 30 minute ACC race after filming this.

Any ideas from where this problem may be coming from? I spent a while searching for someone with the exact same problem without any findings

berarma, 2023-11-24:
I would report it to Proton and see if the problem exists with other wheels too.

Do you have the G923 Playstation edition? Because only that version uses new-lg4ff.

eruizc-dev, 2023-11-24:
Thanks I'll report it to proton to see if anyone else had the same problem, and yes it's the PlayStation edition.

eruizc-dev, 2024-04-09:
UPDATE:

I've noticed this is not exclusive to EA WRC as I was able to replicate this issue in Dirt Rally, Dirt Rally 2.0, and Assetto Corsa Competizione by tweaking the FFB settings in each game. I also tried multiple proton versions across the last few months going from Valve's 7, 8, and 9; and some GE versions too.

The workaround seems to be to find which setting has the biggest impact in this and turn it down, but that ruins the experience in some cases.

berarma, 2024-04-09:
I guess you're referring to the conditional effects: spring, dampen, friction, inertia, etc. Those are generated by the wheel firmware. If there's any functional difference between the G923 and others, it should be in the firmware.

I can't test it myself since I don't have that wheel.

If someone could make a log using ffbtools I could take a look and maybe try to reproduce it.

eruizc-dev, 2024-04-09:
I'll try to gather some logs this week, are there any special configuration or environments you want me to try? I'm guessing linux vs linux-zen, steam flatpak vs native, different proton versions, or different games maybe...

berarma, 2024-04-09:
> I'll try to gather some logs this week, are there any special configuration or environments you want me to try? I'm guessing linux vs linux-zen, steam flatpak vs native, different proton versions, or different games maybe...

A FFB log when the issue happens would give us a reproducible test, at least in the same wheel.

poisotf, 2024-08-19:
I have this wheel, and the same issue.

It can be reproduced by simply playing the 'Friction' effect using ffbtools :
[skip.ffb.gz](https://github.com/user-attachments/files/16665104/skip.ffb.gz)

I tried to compare with the Logitech Windows driver, using qemu, the iRacing WheelCheck.exe, and [this thing](https://github.com/poisotf/FFBInspector) (FFBTest)
It would seem the 'Damper', 'Friction' and 'Inertia' DirectInput effects all play the same 'Damper' firmware effect, and the 'Friction' firmware effect is not used.
here is a USB capture of stacking those 3 effects in 3 slots with FFBTest (without moving the wheel) : 
[g923_DaFrIn_windows_filtered.pcapng.gz](https://github.com/user-attachments/files/16665271/g923_DaFrIn_windows_filtered.pcapng.gz)

berarma, 2024-08-20:
Then, disabling friction in Oversteer should fix the issue. I think friction doesn't work on most Logitech wheels. It's also pretty useless in most games.