berarma/new-lg4ff #87: Cannot get module to load at boot on Fedora 39

Greifent, 2023-12-17:
Hello,
I've recently got the Logitech G29 PS3/PS4, and I cannot get the module to load at boot, using kernel 6.6.6-200
I've tried both with DKMS and the Manual method, and it appears to be loaded correctly:
```
$ sudo lsmod | grep logi
hid_logitech_new       77824  0
hid_logitech_hidpp     77824  0
hid_logitech_dj        40960  0
ff_memless             20480  1 hid_logitech_new
```
and on dmesg:
```
[  212.877013] input: Logitech G29 Driving Force Racing Wheel as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb3/3-4/3-4.3/3-4.3:1.0/0003:046D:C24F.0006/input/input38
[  212.877219] logitech 0003:046D:C24F.0006: input,hidraw5: USB HID v1.11 Joystick [Logitech G29 Driving Force Racing Wheel] on usb-0000:07:00.3-4.3/input0
[  212.877348] logitech 0003:046D:C24F.0006: Force feedback support for Logitech Gaming Wheels (0.4.1)
[  212.877350] logitech 0003:046D:C24F.0006: Hires timer: period = 2 ms
```
Is there something I am missing?
Thanks for the help!

berarma, 2023-12-17:
You need to update the initram disk using using whatever tool Fedora has. Previously it was done automatically by DKMS but this feature got removed and has to be done now manually.

Greifent, 2023-12-17:
Thanks for the response, I tried with `dracut --regenerate-all` as issue #85 suggestes, also tried with `dracut --regenerate-all --force`, after the previous command gave warnings, it finished without output, but after a reboot old drivers were loaded

Greifent, 2023-12-17:
I managed to do it, following this guide:
https://docs.fedoraproject.org/en-US/fedora/latest/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sec-Persistent_Module_Loading

I'll close the issue, thanks!

closed 2023-12-17 by Greifent