berarma/new-lg4ff #93: dmesg outputs only "Device properties not found"

boletales, 2024-03-31:
README says there should be an output like "Force feedback support for ...".

Here are results of some commands:
`uname -a`
```
Linux boletales-thinkcentre 6.1.80-1-MANJARO #1 SMP PREEMPT_DYNAMIC Fri Mar  1 18:09:53 UTC 2024 x86_64 GNU/Linux
```

`sudo dkms status`
```
anbox-ashmem/1: added
anbox-binder/1: added
new-lg4ff/0.4.1, 6.1.80-1-MANJARO, x86_64: installed (original_module exists)
vmware-workstation/17.5.1_23298084, 5.15.150-1-MANJARO, x86_64: installed
vmware-workstation/17.5.1_23298084, 6.1.80-1-MANJARO, x86_64: installed
```

`lsusb`
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 0b39:0001 Omnidirectional Control Technology, Inc. Composite USB PS2 Converter
Bus 001 Device 004: ID 0d8c:0005 C-Media Electronics, Inc. Blue Snowball
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 011: ID 046d:c294 Logitech, Inc. Driving Force
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 002: ID 345f:2131 MACROSILICON UGREEN HDMI Capture
```

`sudo dmesg`
```
[ 4242.911894] usb 1-1.3: USB disconnect, device number 10
[ 4244.013454] usb 1-1.3: new low-speed USB device number 11 using xhci_hcd
[ 4244.381949] usb 1-1.3: New USB device found, idVendor=046d, idProduct=c294, bcdDevice=11.03
[ 4244.381962] usb 1-1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=0
[ 4244.381969] usb 1-1.3: Product: Logitech Driving Force
[ 4244.381975] usb 1-1.3: Manufacturer: Logitech
[ 4244.440042] logitech 0003:046D:C294.000F: fixing up Logitech Driving Force report descriptor
[ 4244.440335] input: Logitech Logitech Driving Force as /devices/pci0000:00/0000:00:02.1/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:046D:C294.000F/input/input37
[ 4244.440668] logitech 0003:046D:C294.000F: input,hidraw4: USB HID v1.00 Joystick [Logitech Logitech Driving Force] on usb-0000:01:00.0-1.3/input0
[ 4244.440689] lg4ff: send_cmd: F8 01 00 00 00 00 00
[ 4255.104478] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104485] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104487] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104488] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104489] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104491] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104492] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104493] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104495] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104496] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104497] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104499] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104500] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104502] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104503] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104504] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104506] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104507] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104508] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104510] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104511] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.104513] logitech 0003:046D:C294.000F: Device properties not found
[ 4255.112480] logitech 0003:046D:C294.000F: Device properties not found
```

`lsmod | grep hid`
```
hid_logitech_hidpp     69632  0
hid_logitech_new       77824  0
hid_logitech_dj        36864  0
ff_memless             20480  1 hid_logitech_new
mac_hid                16384  0
usbhid                 73728  3 hid_logitech_new,hid_logitech_dj,hid_logitech_hidpp
```

boletales, 2024-04-02:
I noticed that DFP (its ID should be c298) was recognized as Driving Force (c294) due to a USB hub.
Everything works correctly after connecting DFP directly to my PC.

closed 2024-04-02 by boletales

berarma, 2024-04-02:
Interesting. I'm glad to see it works now. Thanks for the feedback.