berarma/new-lg4ff #98: Difficult installation guide

doordormeta, 2024-06-15:
Installation guide is too difficult.

Hi, the installation guide is too difficult for me. Why in these steps you do not provide terminal commands?

I get this error (pop-os):
Error! Arguments <module> and <module-version> are not specified.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>

Can you please provide clearer installation guide?

doordormeta, 2024-06-15:
Managed to install with the second method

closed 2024-06-15 by doordormeta

berarma, 2024-06-15:
I don't know which command you run and I can't understand the error.

The DKMS method doesn't specify some commands because they're distro specific or depend on the choosen download method.

I'll try to make it more explicit when a command has to be run but I can't provide the commands for every distro.