berarma/new-lg4ff #100: i’m noob!need some install help:(

chinalhx, 2024-06-29:
I am a newbie in arch linux. Following your method, I successfully installed oversteer+ Thrustmaster T300 RS with hid-tmff2
However, now a new question comes, how do I install new-lg 4ff?
 Is there any one-click installation command suitable for Arch Linux novices? I am willing to buy a cup of coffee as a sponsor@berarma

chinalhx, 2024-06-29:
To be honest, I feel disgusted when I see various commands now, especially when I need to spend time to find a solution
So is there anything I can install with one click? Or copy what you wrote so that it can be used on steamos (archlinux system)? This will make it easier for me to install new-lg4ff

berarma, 2024-06-29:
> To be honest, I feel disgusted when I see various commands now, especially when I need to spend time to find a solution
> So is there anything I can install with one click? Or copy what you wrote so [...]

Do you have a Logitech wheel?

chinalhx, 2024-06-30:
> Do you have a Logitech wheel?

yep！my devices:Steamdeck oiled+Logitech G29(include H shift), Just after this post was posted, I tried to install it myself. I don’t know if it was successful. When I used the’ sudo dmegs ‘command, the output showed loading out of tree.
here’s output link: https://github.com/berarma/new-lg4ff/issues/101

berarma, 2024-06-30:
Answering the original question, there's no one-click command but instructions should be enough.

closed 2024-06-30 by berarma