berarma/new-lg4ff #101: [Requests] seems like install

chinalhx, 2024-06-29:


chinalhx, 2024-06-29:
cd /usr/src
git  clone
cd new-lg4ff
sudo make 
sudo make install
sudo make load
make -C /lib/modules/`uname -r`/build M=$PWD
make -C /lib/modules/`uname -r`/build M=$PWD modules_install
  INSTALL /lib/modules/6.1.52-valve16-1-neptune-61/extra/hid-logitech-new.ko
  SIGN    /lib/modules/6.1.52-valve16-1-neptune-61/extra/hid-logitech-new.ko
At main.c:167:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem
  ZSTD    /lib/modules/6.1.52-valve16-1-neptune-61/extra/hid-logitech-new.ko.zst
  DEPMOD  /lib/modules/6.1.52-valve16-1-neptune-61
depmod -A
rmmod hid-logitech 2> /dev/null || true
rmmod hid-logitech-new 2> /dev/null || true
modprobe hid-logitech-new 

then type: sudo dmesg
hid_logitech_new:loading out of tree module taints kernel
hid_logitech_new:module verification failed:signature and/or required key missing -tainting kerne
is it successful?or missing some type command ?

berarma, 2024-06-30:
It looks good.

closed 2024-06-30 by berarma