berarma/new-lg4ff #103: only autocenter work on G923(ubuntu 20.04)

y1h6b, 2024-07-26:
Hello, respected developer, I am using G923 with this driver on Ubuntu20.04 LTSC(I newlly install this system).
But since I am a novice, the next question may be a bit foolish. Please forgive me.
After installing the driver, I modified the files in sys/bus/hiv/drivers/logitech/XXX: XXXX: XXXX/to change the feedback effect, but I encountered some problems:
1. Only the autocenter file will have an effect after modification, while other files cannot be perceived after modification. Is this a normal situation?
2. I saw in your introduction that the LED will turn on and off with the change of force, but after modifying the autocenter, the force change did not cause the LED to turn on or off. Is this normal?
3. I directly modify the brightness of any LED under LEDS, which can light up the LED. In my opinion, this seems to be a call to the function lg4ff_fled_det_brightness. However, if I comment out the code for LED display in the function lg4ff_timer and directly call the function lg4ff_fled_det_brightness, it will not light up the LED. For example, if I want to light up the third LED (entry ->wdata. led [2], 1), but it does not change.

Thank you for your contribution and for reading my question. Looking forward to your reply!

motolav, 2024-07-26:
Are you using the PlayStation or the Xbox version of the G923, this driver is only required for the PlayStation version.
The Linux kernel includes supports the Xbox version of the wheel after kernel 6.3, this driver will not add support for the Xbox version to older kernels as it uses a different driver.

motolav, 2024-07-26:
There is also a GUI for modifying the wheel settings by the dev of this driver here https://github.com/berarma/oversteer

y1h6b, 2024-07-27:
> Are you using the PlayStation or the Xbox version of the G923, this driver is only required for the PlayStation version. The Linux kernel includes supports the Xbox version of the wheel after kernel [...]

Hello developer, first of all, thank you for reading my question. I carefully checked my packaging box and it says that the compatible hardware is PlayStation and PC, so this should be the PlayStation version, not the Xbox version. 

> There is also a GUI for modifying the wheel settings by the dev of this driver here https://github.com/berarma/oversteer

And I have tried the Oversteer project you sent me before, but it did not work on my G923, so this should once again prove that this is not the Xbox version for me🫡

motolav, 2024-07-27:
For question 1, All the effects on the wheel only work when there is a program sending force feedback data to the wheel. Auto center is a driver/firmware function that forces the wheel back to home even in games so it will always show a change when you adjust the setting.

For question 2, ffb_leds is a monitor of the data the game/program sends the driver so auto centering force won't affect that, `friction_level` is the variable it uses.

motolav, 2024-07-27:
For question 3, I am not a developer so you'd have to ask berama about how to control the LEDs

motolav, 2024-07-27:
> Hello developer, first of all, thank you for reading my question. I carefully checked my packaging box and it says that the compatible hardware is PlayStation and PC, so this should be the PlayStation version, not the Xbox version.

This is the correct driver for your wheel, are you using the wheel with games through proton? There are issues sometimes with proton and racing wheels not working https://github.com/ValveSoftware/Proton/issues/6674