berarma/new-lg4ff #105: Problems with G923 PS4/PC

dasj19, 2024-10-02:
Hello, I installed the 0.4.0 version of the driver on my NixOS system. I also tried a manual install of the code in master branch on a Ubuntu system, The results are the same. I can get the weel recognized by oversteer and gamepad-tool but not by frontends like retroarch or emulationstation (ES-DE).

ES-DE does not recognize a "game pad" device connected.
retroarch displays an error at startup, verbose log follows:
```
[INFO] [udev]: Pad #0 (/dev/input/event7) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event7) supports 16 force feedback effects.
[INFO] [Autoconf]: Logitech G923 Racing Wheel for PlayStation 4 and PC (1133/49766) not configured.
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [DBus]: Suspended screensaver via DBus.
[INFO] [Video]: Found display server: "x11".
```

More details:
Tried on kernel 6.1 and 6.10, no difference.

lsusb output:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 06cb:00a8 Synaptics, Inc. 
Bus 001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 011: ID 046d:c266 Logitech, Inc. G923 Racing Wheel for PlayStation 4 and PC
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 04f2:b685 Chicony Electronics Co., Ltd Chicony USB2.0 Camera
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

last lines of dmesg:
```
[ 6355.664501] usb 1-1: USB disconnect, device number 10
[ 6356.050525] usb 1-1: new full-speed USB device number 11 using xhci_hcd
[ 6356.421417] usb 1-1: New USB device found, idVendor=046d, idProduct=c266, bcdDevice=38.00
[ 6356.421424] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6356.421428] usb 1-1: Product: G923 Racing Wheel for PlayStation 4 and PC
[ 6356.421431] usb 1-1: Manufacturer: Logitech
[ 6356.421434] usb 1-1: SerialNumber: UWOTAMAREJEGYKYLYWEHONOWI
[ 6356.456370] input: Logitech G923 Racing Wheel for PlayStation 4 and PC as /devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-1/1-1:1.0/0003:046D:C266.000D/input/input27
[ 6356.456594] logitech 0003:046D:C266.000D: input,hidraw0: USB HID v1.11 Joystick [Logitech G923 Racing Wheel for PlayStation 4 and PC] on usb-0000:01:00.0-1/input0
[ 6356.456625] logitech 0003:046D:C266.000D: Force feedback support for Logitech Gaming Wheels (0.4.0)
[ 6356.456629] logitech 0003:046D:C266.000D: Hires timer: period = 2 ms
```

evtest reports :
```
Available devices:
/dev/input/event7:	Logitech G923 Racing Wheel for PlayStation 4 and PC
Select the device event number [0-7]: 7
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x46d product 0xc266 version 0x111
Input device name: "Logitech G923 Racing Wheel for PlayStation 4 and PC"
....
```
lsmod | grep logitech
hid_logitech_new       73728  0
ff_memless             20480  1 hid_logitech_new
hid                   172032  5 i2c_hid,usbhid,hid_multitouch,hid_generic,hid_logitech_new

```

I have no idea what I am missing.
I tried connecting the wheel via an USB hub and directly.

Any feedback is greatly appreciated.

berarma, 2024-10-02:
It looks fine. I guess it's a configuration issue in the apps. They're probably looking for something that resembles a gamepad, or maybe they use presets, and they have no preset for a wheel.

There was an SDL app that could create a configuration line so that SDL could use any USB device as an XBox controller. This way, apps wouldn't need a preset.

berarma, 2025-02-05:
I think this issue should be moved to RetroArch and EmulationStation. There's nothing I can do on this side.

closed 2025-02-05 by berarma