berarma/new-lg4ff #109: Error during build on Raspberry PI: hid-lg.c:759:7: error: implicit declaration of function 'hid_is_usb'

Fredobedo, 2024-11-07:
Dear @berarma ,
I have a problem when trying to compile latest release (0.4.1) on my raspberry PI 32 bits.

In short, here is the error message (full log in attachment):
<<
/var/lib/dkms/new-lg4ff/0.4.1/build/hid-lg.c:759:7: error: implicit declaration of function 'hid_is_usb'; did you mean 'hid_parse'? [-Werror=implicit-function-declaration]
  if (!hid_is_usb(hdev))
       ^~~~~~~~~~
>>

I also see that on the last commit of file hid-lg.c, a new function call 'hid_is_usb' (c/o 'Rebase code to Linux kernel 6.1')
And indeed, I don't find any references of this function in all the code.

Could you assist me, please? 

Thank you
Best regards

[make.log](https://github.com/user-attachments/files/17658426/make.log)

berarma, 2024-11-09:
This function is defined in the kernel and declared in the kernel headers. Maybe your kernel is too old. What's your kernel version?

If that's the case you could try an older new-lg4ff version, version 0.4.0 should work.

Fredobedo, 2024-11-14:
thank you for the support, I have indeed succeeded compiling version 0.4.0!

Best regards
Fred

closed 2024-11-14 by Fredobedo