berarma/new-lg4ff #110: Logitech G25 Racing Wheel Issues with new-Ig4ff driver on Fedora

RaulKong898, 2024-11-15:
Hello! 

I'm experiencing several issues with the **new-Ig4ff** driver on Fedora (versions 41 and later) regarding compatibility with my **Logitech G25 Racing Wheel**.

**Incorrect mapping:** Games don't recognize the wheel correctly from the start, and the default mapping is wrong.

 **Clutch issues**
- Many games don't detect the clutch at all, even after manual configuration.
- In other games, the clutch is detected but doesn't function properly. For example, instead of engaging/disengaging the transmission, it might accelerate or brake.
- In some cases, to move forward or backward, I have to press the clutch pedal, shifter, and accelerator pedal simultaneously. This issue occurs constantly, the gear doesn't change, and I'm forced to take my feet off the pedals to make the shift happen.
- **Erratic wheel movement:** In certain games, the wheel moves chaotically and uncontrollably, making it impossible to drive the car properly.

These issues occur in various games, both old and new.

Please give special attention to resolving these issues so that the **Logitech G25 Racing Wheel** works correctly with the **new-Ig4ff** driver on Fedora across all games. 

Thank you for your time and effort!

mksafavi, 2024-12-07:
I think the incorrect mapping is expected as they differ per game. After assigning them manually it should work. 
Although, I've only used it in ACC, AMS2 and beamng.
If you have any of these game, we can compare the behavior your seeing with mine.

maybe the combined_pedals option is enabled on your setup? 

In most games that I tried the pedals axis is reversed. but there's usually an option to reverse them in game.

berarma, 2024-12-08:
I doubt this is related to new-lg4ff. Can you try the games with the built-in module and see if they act the same?