berarma/new-lg4ff #113: Unable to build on Arch 6.12.1

notdexterslab, 2024-11-24:
Kernel: 6.12.1-arch1-1 x86_64 (EndeavourOS)
Wheel: Logitech, Inc. G923 Racing Wheel for PlayStation 4 and PC

All required packages installed, latest `linux-headers` (6.12.1.arch1-1).

Result of running `sudo make` in `/usr/src/new-lg4ff`:
```
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /usr/src/new-lg4ff/hid-lg.o
/usr/src/new-lg4ff/hid-lg.c:936:25: error: initialization of ‘const __u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘const unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} from incompatible pointer type ‘__u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} [-Wincompatible-pointer-types]
  936 |         .report_fixup = lg_report_fixup,
      |                         ^~~~~~~~~~~~~~~
/usr/src/new-lg4ff/hid-lg.c:936:25: note: (near initialization for ‘lg_driver.report_fixup’)
make[3]: *** [scripts/Makefile.build:229: /usr/src/new-lg4ff/hid-lg.o] Error 1
make[2]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /usr/src/new-lg4ff] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
make: *** [Makefile:5: default] Error 2
```

The driver was working fine for me the other day, but when I run `oversteer`, I get "No device available."

berarma, 2024-11-24:
Maybe #112 fixes the issue. You'll need to rebuild the module.

notdexterslab, 2024-11-25:
It seems part of my issues was a broken symlink in /var/lib/dkms/. I deleted the folder, reinstalled `new-lg4ff`, and now everything's working again!

closed 2024-11-25 by notdexterslab