berarma/new-lg4ff #114: Enabling ffb_leds with CONFIG_LEDS_CLASS for all the wheels

mksafavi, 2024-12-07:
Hi.
Is it okay to enable the led support with CONFIG_LEDS_CLASS for all of the supported wheels?

I have the G29 and I noticed that on my distro (NixOS), the module is compiled without the led option. 

I'm going to create a patch to enable it upstream.
wanted to make sure that I'm not breaking it for the other wheels.

berarma, 2024-12-08:
You should ask the NixOS maintainer that disabled it. It's enabled on my system without issues.

mksafavi, 2024-12-08:
> You should ask the NixOS maintainer that disabled it. It's enabled on my system without issues.

I'll ask them. 👍
just wanted to make sure that It's not a G29 exclusive feature that might not work on the other wheels.

berarma, 2024-12-08:
> > You should ask the NixOS maintainer that disabled it. It's enabled on my system without issues.
> 
> I'll ask them. 👍
> just wanted to make sure that It's not a G29 exclusive feature that might not work on the other wheels.

It won't work on wheels without leds.

closed 2025-01-30 by mksafavi